fivdi/pigpio #79: Creating a digital output hangs on the Raspberry Pi 4

nebrius, 2019-07-23:
So here's a strange one for you @fivdi. Creating a digital output on the Raspberry Pi 4 hangs.

Environment:
- Raspberry Pi 4 Model B 2GB (hardware revision b03111)
- Node.js 10.16.0 (installed from NodeSource's PPA)
- OS 2019-06-20-raspbian-buster-lite
- pigpio v1.2.3

Code:
```javascript
const { Gpio } = require('pigpio');

new Gpio(17, {
  mode: Gpio.OUTPUT,
  pullUpDown: Gpio.PUD_OFF
});
console.log('done');
```

When I run this code, "done" is never printed, and a breakpoint set on that line is never triggered. In addition, when I ctrl+c out of the program, I get `sigHandler: Unhandled signal 2, terminating`, which implies something is blocking the event loop to me.

Here's the funny bit: when I run this on a Raspberry Pi 3 Model B+ (hardware revision ) with the exact same code, Node.js version, pigpio version, and OS+updates, it works fine.

Note: I did install Raspi IO first, which means the device was configured for Raspi IO (I2C was enabled and certain parameters were set, etc.).

fivdi, 2019-07-25:
Hey @nebrius, I hope all is well with you.

The 2019-06-20-raspbian-buster and  2019-07-10-raspbian-buster images come with V68 of the pigpio C library preinstalled which doesn't support the Raspberry Pi 4. The pigpio Node.js module depends on the pigpio C library. Run the command `pigpiod -v` to see which version of pigpio C library is installed.

V70 of the pigpio C library provides experimental support for the Raspberry Pi 4. The instructions for installing V70 of the pigpio C library can be found at http://abyz.me.uk/rpi/pigpio/download.html.

Please install V70 of the pigpio C library and give it a try. It should be a lot better than V68.

The corresponding pigpio C library issue is https://github.com/joan2937/pigpio/issues/278

> when I ctrl+c out of the program, I get sigHandler: Unhandled signal 2, terminating, which implies something is blocking the event loop to me.

This message comes from the pigpio C library either [here](https://github.com/joan2937/pigpio/blob/826ab960ede1576b445d914f8f5b7b4d41eda17d/pigpio.c#L5571) or [here](https://github.com/joan2937/pigpio/blob/826ab960ede1576b445d914f8f5b7b4d41eda17d/pigpio.c#L5581).

> Here's the funny bit: when I run this on a Raspberry Pi 3 Model B+ (hardware revision ) with the exact same code, Node.js version, pigpio version, and OS+updates, it works fine.

Compared to previous versions of the Pi, the hardware peripherals on the Pi 4 have changed quite a bit. The pigpio C library therefore needs to be modified accordingly.

> Note: I did install Raspi IO first, which means the device was configured for Raspi IO (I2C was enabled and certain parameters were set, etc.).

This shouldn't be an issue.

cinderblock, 2019-08-07:
I just ran into this issue as well.

Interestingly, this also launched a daemon on port `8888` when running on Pi4. I was very confused until I found this issue.

Will try to update to V70 and give that a shot.

nebrius, 2019-08-07:
Thanks for all the info @fivdi, this is helpful.

> Compared to previous versions of the Pi, the hardware peripherals on the Pi 4 have changed quite a bit. The pigpio C library therefore needs to be modified accordingly.

Interesting, I'll have to dig in when I have some free time (in September 😅)

fivdi, 2019-09-07:
Like the pigpio C library the pigpio Node.js module now has experimental support for the Raspberry Pi 4. Please note that hardware PWM is a little different on the Raspberry Pi 4. For further details see the [New & Updates](https://github.com/fivdi/pigpio#news--updates) section of the README. The integration tests for the Node.js pigpio module all run successfully on a Raspberry Pi 4.

closed 2019-09-07 by fivdi