fivdi/pigpio #80: pigpio install failing with error

stonebraker, 2019-07-26:
My attempts to install pigpio are failing with the error below. Raspbian (Buster). Node 10.16.0 with NPM 6.9.0. Thanks for any help.

make: Entering directory '/home/pi/Desktop/tjbot/dev/tjwave/node_modules/pigpio/build'
  CXX(target) Release/obj.target/pigpio/src/pigpio.o
In file included from ../../nan/nan.h:190,
                 from ../src/pigpio.cc:3:
../../nan/nan_maybe_43_inl.h: In function ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’:
../../nan/nan_maybe_43_inl.h:88:15: error: ‘class v8::Object’ has no member named ‘ForceSet’
   return obj->ForceSet(GetCurrentContext(), key, value, attribs);

fivdi, 2019-07-26:
Which version of pigpio is being installed? It looks like an old version of pigpio that doesn't function with Node 10 is being used.

stonebraker, 2019-07-26:
That was it. Thank you. The package I was installing was referencing 0.4.0. I updated the version in the package, and fortunately the program still worked, which I guess means the interface didn't change too much. Very nice work. And thanks again.

closed 2019-07-26 by stonebraker