fivdi/pigpio #83: Doc - How to disable (servo) PWM ? PWM pin reallocation without reboot ?

Serveurperso, 2019-08-19:
Hello.

This project is fantastic and is essential for small raspberry-pi-only robotics. Sorry for my english.

I need to live reallocate pigpio configuration. I keep all my pigpio objects inside array and it's work.

How to disable PWM for my robot-sleep mode ?
How to reallocate/reset entire pin configuration (how to cleanly delete pigpio objects) ?

Thanks,
Pascal

fivdi, 2019-08-19:
> How to disable PWM for my robot-sleep mode ?

Call `gpioObject.mode(Gpio.INPUT)`

> How to reallocate/reset entire pin configuration (how to cleanly delete pigpio objects) ?

Nothing special needs to be done here. Simply reuse the existing Gpio objects or create new Gpio objects.

Serveurperso, 2019-08-19:
If I call a servoWrite when the pin is always input there is no problem or I get exeption ?

Serveurperso, 2019-08-19:
This is my complete usage of pigpio on robot. "gpioMoteurs[]" is my array of gpio object.

On my websocket re-configuration event :
```
  // All existing to input, kill any residual PWM
  gpioMoteurs.forEach(function(gpio) {
   gpio.mode(GPIO.INPUT);
  });

  // Clear the array of object
  gpioMoteurs = [];

  // Create new array of object from new configuration (number of object can be bigger or smaller than previous configuration!).
  for(let i = 0; i < hard.MOTEURS.length; i++)
   gpioMoteurs[i] = new GPIO(hard.MOTEURS[i].PIN, {mode: GPIO.OUTPUT});
```

On my sleep event :
```
 // Switch all existing to input to disable all PWM and all Mosfets H-bridge
 gpioMoteurs.forEach(function(gpio) {
  gpio.mode(GPIO.INPUT);
 });
```

On my wakeup event :
```
  // Set current configuration of all pin to input
  gpioMoteurs.forEach(function(gpio) {
   gpio.mode(GPIO.OUTPUT);
  });
```

On my robotics binary frame event :
```
 let pwm = some mathematics....

 // Write result from robotics mixing matrix etc...
 gpioMoteurs[i].servoWrite(pwm);
```

fivdi, 2019-08-19:
> If I call a servoWrite when the pin is always input there is no problem or I get exeption ?

I wouldn't expect an exception. As far as I remember the pigpio C library will set the mode to OUTPUT automatically when servoWrite is called.

The code looks OK to me.

Serveurperso, 2019-08-19:
Thanks a lot for this work. This is a must for my project.

If you are curious you can look your work working in realtime here (zero ads)
https://www.vigibot.com/
Select some Robots icon on top bar
Slide your mouse on video : All moves is pigpio:D

It's like a "Continuous integration testing" & automatic update from raw Raspbian Netinstall image on PI Zero / 2 / 3 / 3B+ / 4 !

fivdi, 2019-08-19:
I took a look on a phone so there was no mouse. Looks good. I'll go ahead and close this issue as there isn't a TODO for onoff. Good luck with your project.

closed 2019-08-19 by fivdi