fivdi/pigpio #84: Error: Module version mismatch. Expected 46, got 64.

kueckermann, 2019-09-08:
Hardward: Raspberry Pi 3
OS: Raspbian GNU/Linux 9 (stretch)
Node: 10.16.3

Trying to test the package but it throw the following error upon require.
```
Error: Module version mismatch. Expected 46, got 64.
    at Error (native)
    at Object.Module._extensions..node (module.js:460:18)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    ...
```

kueckermann, 2019-09-08:
Changing the version of Node js changes the expected version.
Node v12.10 -> Error: Module version mismatch. Expected 46, got 72.
Node v8.15.0 -> Error: Module version mismatch. Expected 46, got 57.

fivdi, 2019-09-08:
This error message indicates that pigpio was installed (perhaps a long time ago) with Node.js 4 but a newer version of Node.js was then installed, i.e., Node.js 8, 10 or 12.

fivdi, 2019-09-08:
Please reinstall pigpio after deciding which version of Node.js should be used.

kueckermann, 2019-09-08:
@fivdi I am using pigpio for the first time and have reinstalled after each version change.

kueckermann, 2019-09-08:
So I don't think it would be due to an old install.

fivdi, 2019-09-08:
What's the output of the following command:
```
sudo node -v
```

kueckermann, 2019-09-08:
You are right. v4.2.1
I notice the issue here is that 
node -v  =  v10.16.3
sudo node -v    =  v4.2.1

Any idea why?

kueckermann, 2019-09-08:
Busy reading some forums to fix that sudo issue. Its because nvm is being used and not being used under sudo.

fivdi, 2019-09-08:
The environment isn't setup correctly. There are two versions of Node.js installed. The normal user (I guess user `pi`) and the super user `root` have different PATHs. `pi` and `root` find different versions of Node.js.

fivdi, 2019-09-08:
Try with:
```
sudo $(which node) app-name.js
```

kueckermann, 2019-09-08:
Working now thanks.
Solution I took was to remove node and npm from **/usr/local/bin** and create new links to the nvm versions using:
```
sudo ln -s "$NVM_DIR/versions/node/$(nvm version)/bin/node" "/usr/local/bin/node"
sudo ln -s "$NVM_DIR/versions/node/$(nvm version)/bin/npm" "/usr/local/bin/npm"
```

closed 2019-09-08 by kueckermann