fivdi/pigpio #86: any way to use with the pigpio daemon?

BryanCrotaz, 2019-09-17:
I'm using this from a node.js web server. While I can give the server root access for testing, this isn't a good idea for deployment!

Is there a way to direct this library at the daemon, or is there a recommendation for another way to avoid giving the main UI root access?

fivdi, 2019-09-17:
I'm afraid not. The pigpio C library can only be used by one process at the same time and the pigpio C library requires root privileges.

BryanCrotaz, 2019-09-17:
Which is why the daemon exists. What would need to be added to make this work with the daemon?

fivdi, 2019-09-17:
That's not entirety true. The pigpio C library offers more functionality than the deamon and is extremely fast. The pigpio Node.js module was created to provide access to the additional functionality but also for the direct fast access to hardware peripherals.

fivdi, 2019-09-17:
Consider using a Node.js module that communicates with the daemon or implement code to do this for pwmWrite and pwmRange. I don't want to modify this module to use the daemon rather than the C library.

fivdi, 2019-09-17:
There's no TODO for pigpio here so I'll go ahead and close the issue.

closed 2019-09-17 by fivdi

BryanCrotaz, 2019-09-17:
Todo: support choice of daemon or direct C attachment through a single interface. 

Either with two different classes which have the same interface, or with a Strategy pattern and a switch in the constructor.