fivdi/pigpio #87: packaging project with pigpio with PKG

adoTedo, 2019-10-16:
Hi, 
I'm very happy with your library, I'm trying to package my project with pkg:
https://www.npmjs.com/package/pkg

So that I have just one executable. I'm using a raspberry pi 4. One I try to pack it I got this error, and now I'm lost:
./main: symbol lookup error: /home/pi/projectName/node_modules/pigpio/build/Release/pigpio.node: undefined symbol: _ZN2v811HandleScope12CreateHandleEPNS_8internal24NeverReadOnlySpaceObjectEPNS1_6ObjectE

any advice?

thanks

fivdi, 2019-10-17:
I'm afraid I'm not familiar with pkg and can't provide any help. I'd suggest creating an issue at https://github.com/zeit/pkg/issues. Perhaps someone there knows what the problem is.

Reda1000, 2019-10-25:
Did you also install pigpio when deploying your executable?

Afaik PKG only Packages your node script with all dependencies from npm etc. But not your dependency of pigpio software installed - so adapter / wrapper node-pgpio is bundled but not pigpio.

Reda1000, 2019-10-25:
@adoTedo I just fixed the very same issue you posted:
[image]

By copying the pigpio.node in Release folder into same folder (maybe subfolder also possible) as your binary:
[image]

Or is this breaking the One-Binary-Restriction?

Note: I have seen this .node copying more often in other projects - looks like someone should remember this procedure :)

fivdi, 2019-12-27:
@Reda1000 Thank you for explaining how you resolved the issue. I'm sure the information you provided will be useful for others.

> looks like someone should remember this procedure :)

Unfortunately I don't know enough about PKG to document what needs to be done here correctly.

I'll go ahead and close this issues as it's not directly related to pigpio.

closed 2019-12-27 by fivdi