fivdi/pigpio #88: Run stepper motor for a precise number of steps.

philipp-cloudproduce, 2019-11-21:
Thanks for making this package. It is really great. I am unable to figure out how to do something specific that I am after though. I would like to be able to turn stepper motor for a precise number of steps. Currently the only thing I can see that I can do is to execute pwmWrite and then set a javascript timer to go off after a specified number of milliseconds and then turn off the pwm. But this seems very imprecise to me. Is there a way that I can set a pwm frequency and duty cycle and then a duration in microseconds? Or even set a number of steps as an input? If this is not currently supported, is it something that could be added to the API? It would make working with stepper motors precisely a lot easier. Thanks!

fivdi, 2019-11-22:
Is it possible to achieve what you would like to achieve by doing something similar to what [this Python program](https://github.com/stripcode/pigpio-stepper-motor/blob/master/PigpioStepperMotor.py) does in JavaScript?

>Is there a way that I can set a pwm frequency and duty cycle and then a duration in microseconds? Or even set a number of steps as an input?

If the pigpio Node.js module provided bindings for the [Waves](http://abyz.me.uk/rpi/pigpio/cif.html) functions available in the pigpio C library then this would be possible. However, the pigpio Node.js module doesn't provide bindings for the Waves functions available in the pigpio C library. See #38.

fivdi, 2019-12-07:
Closing in favour of #38

closed 2019-12-07 by fivdi