fivdi/pigpio #89: Digital Pulse

louisvangeldrop, 2019-11-22:
Espruino.com supports the following digitalPulse function:

Call type:
function digitalPulse(pin, value, time)
Parameters
pin - The pin to use

value - Whether to pulse high (true) or low (false)

time - A time in milliseconds, or an array of times (in which case a square wave will be output starting with a pulse of 'value')

Description
Pulse the pin with the value for the given time in milliseconds. It uses a hardware timer to produce accurate pulses, and returns immediately (before the pulse has finished). Use digitalPulse(A0,1,0) to wait until a previous pulse has finished.

eg. digitalPulse(A0,1,5); pulses A0 high for 5ms. digitalPulse(A0,1,[5,2,4]); pulses A0 high for 5ms, low for 2ms, and high for 4ms

Note: if you didn't call pinMode beforehand then this function will also reset pin's state to "output"

digitalPulse is for SHORT pulses that need to be very accurate. If you're doing anything over a few milliseconds, use setTimeout instead.

My question is: how do I get the same functionality using piGpio?

fivdi, 2019-11-22:
If the pigpio Node.js module provided bindings for the [Waves](http://abyz.me.uk/rpi/pigpio/cif.html) functions available in the pigpio C library then I think this would be possible. However, the pigpio Node.js module doesn't provide bindings for the Waves functions available in the pigpio C library.

This is a duplicate of #38

louisvangeldrop, 2019-11-26:
I will have a look at pigpio-client. The package supports the Wave functionality

fivdi, 2019-12-07:
Closing in favour of #38

closed 2019-12-07 by fivdi