fivdi/pigpio #90: configureClock go above 10 microSeonds sample rate?

bodtx, 2019-11-23:
according to the [doc](https://github.com/fivdi/pigpio/blob/master/doc/configuration.md#configureclockmicroseconds-peripheral) you cannot go below 10 microseconds

as I can see on my oscilloscope I cannot go below approximatly 1/(2.4×10⁻⁶) = 416666
416Khz with pigpio.configureClock(10, pigpio.CLOCK_PCM).

Why cannot we go above 10 on the first parameter?
how could I go to 20kHz (to control motors)?

I have already one motor on the hardware pwm pin.

fivdi, 2019-11-24:
What frequency does the oscilloscope see on GPIO17 if the below program is run. It should see approximately 20kHz.
```js
const pigpio = require('pigpio');
const Gpio = pigpio.Gpio;

pigpio.configureClock(1, pigpio.CLOCK_PCM);

const led = new Gpio(17, {mode: Gpio.OUTPUT});

led.pwmFrequency(20000);
led.pwmRange(50);
led.pwmWrite(25);

setTimeout(_ => {}, 1e6);
```

fivdi, 2019-12-07:
To the best of my knowledge the information needed has been provided here so I'll go ahead and close the issue,

closed 2019-12-07 by fivdi

bodtx, 2019-12-07:
Sorry for responding late, after testing to the oscilloscope, it works! 🏅

When monitoring the cpu it was at 25% approximately as the doc says. 
I've tried to lower the `configureClock` to 5 or so but it was impossible to reach the 20KHz.

maybe you should add a word in the doc about `pwmFrequency pwmRange`  ?

fivdi, 2019-12-07:
> Sorry for responding late, after testing to the oscilloscope, it works! 🏅

No problem.

> maybe you should add a word in the doc about `pwmFrequency pwmRange`  ?

The documentation for `pwmFrequency` is [here](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#pwmfrequencyfrequency). As can be seen in the table, a frequency of 20kHz is possible when the sampling rate is 1us or 2us. The table also shows that the maximum frequency at a sampling rate of 5us is 8kHz.

The documentation of `pwmRange` is [here](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#pwmrangerange).

What can be added to the documentation to improve things?