fivdi/pigpio #92: PIGPIO release Version 73 to support 512 I2C devices

yboujraf, 2020-01-02:
Dear @fivdi 

Few hours ago the repository released a new version of PIGPIO

https://github.com/joan2937/pigpio/releases/tag/V73

Do you think your repository could sync the new release  and get the new features ?

Best Regards,
Youssef

fivdi, 2020-01-02:
As mentioned in the [pigpio Node.js module readme](https://github.com/fivdi/pigpio#pigpio), the pigpio Node.js module is a wrapper for the [pigpio C library](https://github.com/joan2937/pigpio). When the pigpio Node.js module is installed with the command `npm install pigpio`, the C++ code in the pigpio Node.js module is compiled and linked with the shared pigpio C object library called libpigpio.so from [pigpio C library](https://github.com/joan2937/pigpio). In other words, a dynamically linked shared library is used (libpigpio.so). A statically linked library that becomes part of the pigpio Node.js module is not used.

This means that when a new version of the [pigpio C library](https://github.com/joan2937/pigpio) is released it can simply be installed and the [pigpio Node.js module](https://github.com/fivdi/pigpio#pigpio) will function correctly if there are no breaking changes in the pigpio C library.

There is no TODO for the pigpio Node.js module here. Simply install V73 of the pigpio C library and everything should function correctly.

yboujraf, 2020-01-02:
Dear @fivdi 

On which folder I need to copy the new release in your repository ?

Best Regards
Youssef

fivdi, 2020-01-02:
> On which folder I need to copy the new release in your repository ?

You don't need to copy anything anywhere. You need to download and install V73 of the pigpio C library as described [here](http://abyz.me.uk/rpi/pigpio/download.html).

fivdi, 2020-01-04:
@yboujraf I successfully upgraded from V71 to V73 of the pigpio C library using the instructions provided [here](http://abyz.me.uk/rpi/pigpio/download.html) and the previously installed version of the pigpio Node.js module functioned correctly before and after upgrading to V73 of the pigpio C library.

Note that since the release of the first version of the pigpio Node.js module in 2015 there have never been any breaking changes to the pigpio C library that resulted in the pigpio Node.js module breaking. The pigpio Node.js module and pigpio C library have always worked well together.

There is no TODO for this repository here so I'll go ahead and close this issue.

closed 2020-01-04 by fivdi

yboujraf, 2020-01-04:
Congratulations @fivdi

Do you updated your repository ? And i can clone it ?

Other just follow your instruction in your last email

I appreciate to have taken consideration of my issue

Best regards
Youssef

fivdi, 2020-01-04:
>Do you updated your repository ?

No, the pigpio Node.js module was not updated as nothing needed to be updated.

>And i can clone it ?

Yes.

>Other just follow your instruction in your last email

Correct.

>I appreciate to have taken consideration of my issue

You're welcome.