fivdi/pigpio #93: PIGPIO - Motorized fader - L293D - confusing

yboujraf, 2020-01-06:
Dear @fivdi 

I received yesterday from express my new toys, motorized fader 10kOhms with motorized fader.

I connect my raspbery PI4 with the L293D as recommanded by the manufacturer and Added a 9v DC for the VCC2 to supply the H-bridge and followed the true table.

First again mismatch between the id of the GPIO to assign in the library and pin layout of the connector.
I need to use the BCM id but for cabling I need to use pin layout of the connector ;-) spaghetti.

Finally, I used the PIGPIO libray ;-) 

But to see my fader going down and then going up was exciting :-(

```
'use strict';

const Gpio = require('../pigpio').Gpio;

const wait = function (t) {
    return new Promise(resolve => {
        setTimeout(resolve, t);
    });
}

const motora = new Gpio(26, { mode: Gpio.OUTPUT });
const motorb = new Gpio(19, { mode: Gpio.OUTPUT });
let dutyCycle = 10;

Promise.resolve()
    .then(() => {
        motora.digitalWrite(0);
        motorb.digitalWrite(1);
    })
    .then(() => {
        const timeout = 1000;
        console.log(`Waiting ${timeout} ms...`)
        return wait(timeout);
    })
    .then(() => {
        motora.digitalWrite(1);
        motorb.digitalWrite(0);
    })
    .then(() => {
        const timeout = 1000;
        console.log(`Waiting ${timeout} ms...`)
        return wait(timeout);
    })
    .then(() => {
        motora.digitalWrite(0);
        motorb.digitalWrite(0);
    })

    .catch((err) => console.error(`ERROR`, err));
```
Now to close the control It is missing my 12 bits AD converter to read the value of the position but not possible till Thursday :-(

Then I expect to use PWM, to say I have 4096 positions ( 12 bits adc) then I need 4096 step to parse my range (min - max) but the only surprise I got is my motorized fader vibrated and finally an error message from the library :-(

Probably I am not using correctly the library because I see PWM and hardwarepwm. I see servo example but in my case I can't do nothing with that.

The true table is clear :

const motora = new Gpio(26, { mode: Gpio.OUTPUT });
const motorb = new Gpio(19, { mode: Gpio.OUTPUT });

motara = 1
motorb =0 
=> going up

motora =0
motorb = 1
=> going down

motora =0
motorb = 0
=> stop

So now, what the next step :

- to calibrate :

+ the fader, I have also seen if fader is off, the resistance is = 164 ohms not 0 !!!!
If open resistance is = 9.84 kohms not 10kohms !!!!
 + what about the pigpio library, do I need to use PWM with L293D or use simply like the code above and read the value of the ADC and stop when get the position.
=> No precision in this case.

If you have any clear idea about my issue it will be great

I hope you could help me My Best Dear

Youssef

fivdi, 2020-01-06:
>Then I expect to use PWM, to say I have 4096 positions ( 12 bits adc) then I need 4096 step to parse my range (min - max) but the only surprise I got is my motorized fader vibrated and finally an error message from the library :-(

You'll need to debug your code to figure out why the vibration/error is occurring.

>the fader, I have also seen if fader is off, the resistance is = 164 ohms not 0 !!!!
If open resistance is = 9.84 kohms not 10kohms !!!!

The range 164 ohms to 9.84 kohms will have to be mapped to the range 0 to 4095.

>what about the pigpio library, do I need to use PWM with L293D or use simply like the code above and read the value of the ADC and stop when get the position.

If it's necessary to control the motor speed, PWM will be needed. If it's not necessary to control the motor speed PWM will not be needed.

There doesn't appear to be an issue with the pigpio Node.js library here.

yboujraf, 2020-01-06:
Dear @fivdi 

That's right I deed some mathematics.

I used 2 points (x1,Y1) and (X2,Y2) to calculate the function when I get the value from the ADC I will have the correct information.

y = m.x + b

My daughter helps me to figure out this ;-)

I tried the PWM example in the repository but nothing except this error 

```
'use strict';

const Gpio = require('../pigpio').Gpio;

//const Gpio = pigpio.Gpio;
const motora = new Gpio(26, { mode: Gpio.OUTPUT });
const motorb = new Gpio(19, { mode: Gpio.OUTPUT });

let pulseWidth = 1000;
let increment = 0;
let iv;

process.on('SIGINT', () => {
  motora.digitalWrite(0);
  motorb.digitalWrite(0);
  clearInterval(iv);
  console.log('Terminating...');
});

motora.digitalWrite(0);
motorb.digitalWrite(0);

setInterval(() => {
  console.log("in intervall loop");
  increment++;
  if (increment <= 25) {
    //motorb.digitalWrite(0);
    motora.servoWrite(pulseWidth);
    console.log("increment : " + increment + " - pulsewidth : " + pulseWidth);
  }
  else {
    process.exit(1)
  }

}, 1000);

```

It is very slow.

If I change the pulsewidth to 10000. I gor error message

```
pi@byraspberry:~/share/Projects/_PIGPIO $ sudo node ./pigpio-master/example/motorized-fader-pwm.js 
in intervall loop
2020-01-06 23:49:06 gpioServo: gpio 26, bad pulsewidth (10000)
/home/pi/share/Projects/_PIGPIO/pigpio-master/pigpio.js:134
    pigpio.gpioServo(this.gpio, +pulseWidth);
           ^

Error: pigpio error -7 in gpioServo
    at Gpio.servoWrite (/home/pi/share/Projects/_PIGPIO/pigpio-master/pigpio.js:134:12)
    at Timeout._onTimeout (/home/pi/share/Projects/_PIGPIO/pigpio-master/example/motorized-fader-pwm.js:28:12)
    at listOnTimeout (internal/timers.js:537:17)
    at processTimers (internal/timers.js:481:7)
pi@byraspberry:~/share/Projects/_PIGPIO $

```

I am lost Dear @fivdi 

Best Regards
Youssef

fivdi, 2020-01-07:
Please take a look at the documentation for [servoWrite](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#servowritepulsewidth) to see why an error is thrown.

The motor on the motorized fader is almost certainly not a servo motor so `servoWrite` will not be the correct method to use. You'll need to check the datasheet for the motorized fader to see what type of motor it has. It's most likely a dc motor so the `pwmWrite` can be used to control the speed.

fivdi, 2020-01-07:
The error codes from V73 of the pigpio C library can be found [here](https://github.com/joan2937/pigpio/blob/V73/pigpio.h#L6319-L6469).

In the "bad pulsewidth (10000)" stack trace above we can also see "Error: pigpio error -7 in gpioServo":
```
2020-01-06 23:49:06 gpioServo: gpio 26, bad pulsewidth (10000)
/home/pi/share/Projects/_PIGPIO/pigpio-master/pigpio.js:134
    pigpio.gpioServo(this.gpio, +pulseWidth);
           ^

Error: pigpio error -7 in gpioServo
```
Error code -7 can be found [here](https://github.com/joan2937/pigpio/blob/V73/pigpio.h#L6327) and is:
```
#define PI_BAD_PULSEWIDTH    -7 // pulsewidth not 0 or 500-2500
```

yboujraf, 2020-01-07:
Dear @fivdi 

You are the king of your repository.
I appreciate to share with me all your knowledge.

When you are in Brussels, let me know to share chocolate and beer ;-)

Best Regards,
Youssef

yboujraf, 2020-01-08:
Dear @fivdi 

## Raspberry PI4 BCM2711

### Mode and pin layout of each IO

I don't know if you would like to add this in your documentation because ALT0,1,2,3,4,5 it was unclear for me the I/O with raspberry PI4 because most of the time we found library for PI3.

I found this link : https://elinux.org/RPi_BCM2711_GPIOs or this coming from raspberry page 9 & 10

[rpi_DATA_2711_1p0_preliminary.pdf](https://github.com/fivdi/pigpio/files/4033247/rpi_DATA_2711_1p0_preliminary.pdf)

Now it is clear for me how to mange all I/O of my raspberry.

## My use case - controlling the DC motor to move to a defined position

### First I need calibration.

As told in my previous comments, the motorized fader is not really 0 till 10kOhm but 164 ohm till 9.8kohms. That means I need to use the value received by the Analog to digital converter to know the min and max

I have an analog to digital converter 12 bits (mcp3208) to read the position of the fader.
I calculate the formula because I know two points min(value from ADC for 164ohm, -128dB) and max(value from ADC for 9.8kohms,+15 dB)

I am able to calculate the equation by calculating m and b.

y= m.x + b and i will skip the rest

### How to control the position of the fader with precision with my DC motor when receiving the value -14dB from the audio console and before I was at -128dB ?

I am using an L293D Dual H-Bridge to manage the DC motor. I am able to manage the direction forward or backward, release the motor then the user could change manually the position of the fader and the ADC read the value.

- For the direction of the fader I need 2x GPO to control the direction and to release the motor (using as manually)
- To dis/enable the movement of the fader I have a GPO too

But finally, what I need is if I have an analog to digital converter 12 bits (4096 positions) without interrupt , I need to poll the value but if I have 56 faders :-(, 

=> I would like to calculate the pulse I need to send to the Dis/enable GPO with the direction for moving forward or backward to the position (dB) with precision  or following the value received from an audio console without to see the fader shacking because resolution of the ADC 

Any suggestion My Dear

Best Regards
Youssef

fivdi, 2020-01-08:
The pigpio Node.js module isn't the correct tool for controlling 56 faders with PWM. It can only control the GPIOs that are on the Raspberry Pi GPIO header.

yboujraf, 2020-01-08:
Dear @fivdi 

To control the 56 faders we found a very simple solution and the manufacturer send us today all the information.

To control the DC motor they used D register to control the direction and a push-pull amplifier with transistor  (old fashion) now L293D could do everything.

The last question is how to control the position of the fader with precision with your library ;-)
I though using D register with a clock but knowing I have 4096 positions with 12 bits adc.

My black hair changed to white now. In arduino I followed youtube and it is done in 5 lines but with my proud raspberry I am lost.

Best Regards
Youssef

fivdi, 2020-01-09:
When you say D register do you mean a shift register? I don't understand how it's possible to control the speed of 56 DC motors with PWM using a D register / shift register.

yboujraf, 2020-01-09:
Dear @fivdi 

I received the drawing and they are using :

[image]

[image]

for each motorized fader

then 

[image]

[image]

### From my side

- L293D
- 3x GPO and that's all

But not able to move the fader at a defined position.

I think my AD Converter is missing and stoled from my postbox yesterday (Amazon is looking for my delivery know :-().

I think this is what it is missing to close my issue. But need to use PWM to define the speed of the fader.

### Check those videos 

https://www.youtube.com/watch?v=r12wG7ebvnY
https://pastebin.com/bgF5uWBa

https://motorizedfader.wordpress.com/

Best Regards,
Youssef

fivdi, 2020-01-09:
@yboujraf Please take a look at [this](https://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Header-with-Photo.png) diagram of the Raspberry Pi GPIO header. It has 26 GPIO pins that can be used for PWM, 28 if ID_SC and ID_SD are also used as GPIOs.

If every GPIO available on the Raspberry Pi GPIO header is used for PWM a maximum of 28 PWM pins would be available. You need 56 PWM pins. The circuit diagrams posted above do not change this.

The pigpio Node.js modules is not suitable for what you want to achieve.

fivdi, 2020-01-12:
I'll go ahead and close this issue as there is no todo for pigpio.

closed 2020-01-12 by fivdi