fivdi/pigpio #94: Control Servo to Specific Angles

bradocchs, 2020-01-08:
Hello,

I'm trying to control a servo to specific angles, but I'm having trouble figuring out how. Is there any example code on how to move a servo to a specific angle and then hold that position?

I've tried different things with both servoWrite and pwmWrite with no luck.

Thanks!!

Brad

fivdi, 2020-01-09:
The angle will have to be converted to microseconds with code something like this:
```js
let angle = 180;
let min = 1000;
let max = 2000;
let microseconds = (min + (angle / 180) * (max - min));
console.log(microseconds);
```

fivdi, 2020-01-09:
I think the required information has been provided so I'll go ahead and close this issue.

closed 2020-01-09 by fivdi

bradocchs, 2020-01-10:
When I do this it just spins CW or CCW (depending on microsecond value) until the next command or end of program. I've tried with and without setInterval. I'm using a Parallax continuous rotation servo for testing on GPIO 10 with a RPI 4. I was hoping to be able to rotate 90 degrees and stop. It also seems to coast instead of brake. I may end up using an Arduino which works fine.

fivdi, 2020-01-10:
If it's a continuous servo set the angle to 90 degrees (approx. 1500 microseconds) to stop and hold the servo.

bradocchs, 2020-01-11:
Quick update: I had to adjust the pot on the continuous to get it to stop and then I ended up switching to a standard servo and everything works great!