fivdi/pigpio #97: Error: pigpio error -123 in gpioSetISRFunc

MoffKalast, 2020-01-12:
So far I've been a big fan of this library since it's always worked flawlessly for me with pwm outputs. Today I needed a button input though and it appears to crash on setting up the interrupts.

```
/home/ubuntu/ros/roscar/src/driver/node_modules/pigpio/pigpio.js:148
    pigpio.gpioSetISRFunc(this.gpio, +edge, +timeout, handler);
           ^

Error: pigpio error -123 in gpioSetISRFunc
    at Gpio.enableInterrupt (/home/ubuntu/ros/roscar/src/driver/node_modules/pigpio/pigpio.js:148:12)
    at new Gpio (/home/ubuntu/ros/roscar/src/driver/node_modules/pigpio/pigpio.js:57:12)
    at Object.<anonymous> (/home/ubuntu/ros/roscar/src/driver/scripts/buttontest.js:7:14)
    at Module._compile (internal/modules/cjs/loader.js:959:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:995:10)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Function.Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module.runMain (internal/modules/cjs/loader.js:1047:10)
    at internal/main/run_main_module.js:17:11

```

Test code:

```
const pigpio = require('pigpio');
const Gpio = pigpio.Gpio;

var button = new Gpio(25, {
	mode: Gpio.INPUT,
	pullUpDown: Gpio.PUD_DOWN,
	edge: Gpio.EITHER_EDGE
});

var buttonval = false;
button.on('interrupt', (level) => {
	buttonval = !buttonval;

	if(buttonval)
		console.log("button pressed");
	else
		console.log("button released");
});
```

Running Ubuntu 16.04 on Pi 4, with pigpiod version 72 and node v12.13.1.

Any ideas? Thanks.

fivdi, 2020-01-12:
The code posted above works as expected for me on a Raspberry P 4 with Raspbian 2019-09-26, Node.js v12.14.0, v3.0.0 of the pigpio Node.js module and V73 of the pigpio C library.

[Error -123](https://github.com/joan2937/pigpio/blob/V73/pigpio.h#L6446) means "bad ISR initialisation" and occurs [somewhere here](https://github.com/joan2937/pigpio/blob/V73/pigpio.c#L11552-L11658) in the pigpio C library.

There appears to be something special about GPIO25 on your system.

What happens if the following commands are run manually form the command line on the Raspberry Pi, are there errors?

```
sudo -s
cd /sys/class/gpio/
echo 25 > export 
cd gpio25
echo in > direction 
echo both > edge 
cd /sys/class/gpio/
echo 25 > unexport 
```

MoffKalast, 2020-01-12:
Well...

```
root@roscar:/sys/class/gpio# echo 25 > export
bash: echo: write error: Device or resource busy
root@roscar:/sys/class/gpio# cd gpio25
bash: cd: gpio25: No such file or directory
root@roscar:/sys/class/gpio# ls
export  gpiochip0  gpiochip100  gpiochip504  unexport
```

fivdi, 2020-01-12:
Something has GPIO25 reserved.
Is additional hardware and the corresponding drivers installed on the system that could be reserving GPIO25?
Are additional device tree overlays loaded that could be reserving GPIO25?

fivdi, 2020-01-12:
>I guess we can say the C version of pigpio doesn't really work perfectly with Ubuntu in the first place.

I don't know if this is the case as I've never tried it.

fivdi, 2020-01-12:
>I can do a workaround with on('alert') which works, but this could probably use patching if it's straight forward to do so.

Working around the issue using alerts probably isn't a good idea as whatever has GPIO25 reserved may mess thing up at some point, especially if whatever has GPIO25 reserved has it reserved as an output...

MoffKalast, 2020-01-12:
Hold on, I've just remembered that I've actually compiled pigpio from source on this machine, so the files may not be in standard locations. Could that be the problem for your debug tests?

>Are additional device tree overlays loaded that could be reserving GPIO25?

Well I'm running a handful of i2c modules, but that's about it (and none are active at this time). Can't imagine what it could be.

fivdi, 2020-01-12:
>Hold on, I've just remembered that I've actually compiled pigpio from source on this machine, so the files may not be in standard locations. Could that be the problem for your debug tests?

I wouldn't think so. You are seeing error -123 with the JavaScript program posted above and you are seeing "Device or resource busy" errors when doing the same on the command line. This is consistent and indicates that everything is ok with the pigpio C library.

> Well I'm running a handful of i2c modules, but that's about it (and none are active at this time). Can't imagine what it could be.

Unfortunately, you'll have to figure out what's reserving GPIO25. Or maybe try with a different GPIO?

MoffKalast, 2020-01-12:
Hmm, right. Checking other pins doesn't give me any errors. I guess I'll move the pin then. ¯\\\_(ツ)_/¯

closed 2020-01-12 by MoffKalast

fivdi, 2020-01-12:
#42 was a similar issue but in that case it was with GPIO4 which was being reserved for a 1-Wire interface.

@MoffKalast Good Luck :smile:

TMagerat, 2024-07-31:
> The code posted above works as expected for me on a Raspberry P 4 with Raspbian 2019-09-26, Node.js v12.14.0, v3.0.0 of the pigpio Node.js module and V73 of the pigpio C library.
>
> [Error [...]

Hello,

I don't usually use Github so excuse me if I'm not using it in the right way ...
However, I have the same error as @MoffKalast on my raspberry pi model 3 but when I do the recommended command, I have an error "Invalid argument" just after: echo 24 > export
(I tried to solve the problem by using other gpio ( 17, 23 and 24 ) but it doesn't work.
Can you help me ?

Thanks in advance