fivdi/pigpio #98: Update to v74 c-library

patonz, 2020-01-28:
Too many times i got the address "already in use" from something. seems to be resolved in the last rel of the c library. need to be updated

fivdi, 2020-01-29:
> Update to v74 c-library

There doesn't appear to be a v74 of the pigpio C library, at least not at the time of writing, see [here](https://github.com/joan2937/pigpio/releases).

The pigpio Node.js module doesn't actually install the pigpio C library. The pigpio Node.js module assumes that the pigpio C library has already been installed. For further information see the [installation section of the readme](https://github.com/fivdi/pigpio#installation).

> Too many times i got the address "already in use" from something.

Did the error message that was displayed complain about port 8888 already being in use with an error message like this?
```
initInitialise: bind to port 8888 failed (Address already in use)
```
If this is what happened, pigpio can be told to use a different port number by calling [configureSocketPort](https://github.com/fivdi/pigpio/blob/master/doc/configuration.md#configuresocketportport). Alternatively, pigpio can be told not to use a socket by calling [configureInterfaces](https://github.com/fivdi/pigpio/blob/master/doc/configuration.md#configureinterfacesifflags).

> seems to be resolved in the last rel of the c library. need to be updated

As mentioned above, the pigpio Node.js module doesn't actually install the pigpio C library so there is no TODO for the pigpio Node.js module here.

fivdi, 2020-02-08:
Raspbian Buster 2020-02-05 comes with V71 of the pigpio C library preinstalled. The pigpio Node.js module works as expected on Raspbian Buster 2020-02-05.

closed 2020-02-08 by fivdi