fivdi/pigpio #99: Nothing happens when running example code

proateen, 2020-02-12:
I have wired my HC-SR04 sensor to my Rasp Pi. I have installed Node, NPM and Pigpio.
I run `sudo node distance-hc-sr04.js` and literally nothing happens. A new line gets created and that's it. I have no idea what the cause of it is. I have double checked the wiring already.

fivdi, 2020-02-12:
Can you post a photo of the circuit that clearly shows the HC-SR04 sensor, the Raspberry Pi GPIO header, and the wires connecting the sensor to the header please?

fivdi, 2020-02-12:
I just tried `distance-hc-sr04.js` with the circuit from the [readme](https://github.com/fivdi/pigpio#measure-distance-with-a-hc-sr04-ultrasonic-sensor) on a Raspberry Pi 3 B and it functioned as expected.

proateen, 2020-02-12:
Thank you for your swift response. I've been trying to diagnose this issue for the past week :/ Please see the pictures attached. I'm using a Raspberry Pi 3B. Something strange happened after I posted the issue. After I ran the command, I left the console open for about half an hour and eventually I got a reading, but the reading looks really off.

Those resistors are 1k each.

[image]

[image]

[image]

[image]

fivdi, 2020-02-12:
A HC-SR04 needs to be powered with 5V not 3.3V. The photos show that the red wire on the GPIO header is connected to 3.3V (pin 1 on the GPIO header,) it should be connected to 5V (pin 2 on the GPIO header.)

See [here](https://github.com/fivdi/pigpio#measure-distance-with-a-hc-sr04-ultrasonic-sensor).

It would be better to use a 470 ohm and 330 ohm resistor but with a bit of luck two 1k resistors will work.

fivdi, 2020-02-13:
TRIG and ECHO are also connected incorrectly.

GPIO23 (pin 16 on the Raspberry Pi GPIO header) should be connected to TRIG on the HC-SR04 but photos show it's connected to ECHO.

GPIO24 (pin 18 on the Raspberry Pi GPIO header) should be connected to ECHO on the HC-SR04 but the photos show it's connected to TRIG.

proateen, 2020-02-13:
Correct, sir! Got it working in the end! Thank you!
Re-wired it and it started working.

closed 2020-02-13 by proateen