fivdi/pigpio #100: Can I read an output pin?

BobSegrest, 2020-02-18:
I am using a raspberry pi in a device control application.  I have a digital output line that drives a relay, but I also need to read the current state of the output line to drive a status indicator.  If I define the pin as an output, is there any reason I can't read the output state or value?

My interface allow multiple connections, so I don't want to simply set a value and keep track of it for status display.  I also need to be able to read the state to know that another operator has changed the value.

Is there a reason why I can't read an output value?  

Bob Segrest

fivdi, 2020-02-18:
> I am using a raspberry pi in a device control application. I have a digital output line that drives a relay, but I also need to read the current state of the output line to drive a status indicator. If I define the pin as an output, is there any reason I can't read the output state or value?

It's possible to read the state of a pin that is defined as an output.

> My interface allow multiple connections, so I don't want to simply set a value and keep track of it for status display. I also need to be able to read the state to know that another operator has changed the value.

It's not necessary to keep track of the value of the pin. As mentioned above, it's possible to read the state of a pin that is defined as an output.

>Is there a reason why I can't read an output value?

I'm not aware of anything that prevents the value of an output pin from being read.

fivdi, 2020-02-18:
Take a look at the [gpio-info](https://github.com/fivdi/pigpio/blob/master/example/gpio-info.js) example.

BobSegrest, 2020-02-18:
Thank you for the quick responses!

I started using raspi-gpio and it does not appear able to read a pin configured for output...

Bob

closed 2020-02-18 by BobSegrest