fivdi/pigpio #101: Enhanced : Processing a GPI event (rasing, falling or edge) by interrupt

yboujraf, 2020-02-22:
Dear @fivdi 

I have a use case and I 'd like to check if it could be an enhanced functionality in your repository.

## GPI - Interrupt enabled (falling , rising or edge) processing

[image]

[image]

### Use case

1. I receive a pulse from a device
2. I need to detect the pulse. that means I need to enable GPI with interrupt (edge)
3. the rising edge could be delayed. Tin = x ms but could be 0
4. the falling edge could be delayed. Tout = x ms but could be 0
5. At the falling edge, I need to delay it. P2 = x ms but could be 0.

Finally I need to have :

- Delay the rising edge => Tin >= 0 ms
- Delay the falling edge => Tout + P2 >= 0 ms

I need this functionality per GPI.

What do you think ?

Best Regards
Youssef

fivdi, 2020-02-22:
What have you tried and what were the problems?

fivdi, 2020-02-29:
I don't see any reason to enhance pigpio here.

closed 2020-02-29 by fivdi