fivdi/pigpio #102: motor.servoWrite(0) not working

Janaka-Steph, 2020-02-22:
Hello,

I would like my Tower Pro Micro Servo 9g to stop applying force and keeping the position after the action is done. I tried `motor.servoWrite(0)`, `motor.pullUpDown(Gpio.PUD_OFF)`, `motor.pullUpDown(Gpio.PUD_DOWN)`, `motor.digitalWrite(0)` but nothing works. I don't see what I am doing wrong.

`
const motor1 = new Gpio(26, {mode: Gpio.OUTPUT})
const motor2 = new Gpio(19, {mode: Gpio.OUTPUT})

const getMotor = (pin: number) => {
  switch (pin) {
    case 1: {
      return motor1
    }
    case 2: {
      return motor2
    }
}

async function activateServo(pin: number) {
  let pulseWidth = 800
  while (pulseWidth !== openPosition + 100) {
    getMotor(pin).servoWrite(pulseWidth)
    await sleep(30)
    pulseWidth += increment
  }
  getMotor(pin)?.servoWrite(0)
`

fivdi, 2020-02-22:
Calling `motor.servoWrite(0)` should do it.

The program posted above is not a complete program and contains errors. For example, the following line contains a '?'

```js
  getMotor(pin)?.servoWrite(0)
```

Please post a complete and minimal program that can be used to reproduce the error.

Janaka-Steph, 2020-02-22:
I have edited the example above. This is basically the code I have but with more servos. The "?" Typescript, I have removed it.

fivdi, 2020-02-22:
Did you run the program posted above to verify that it reproduces the error? I don't think so as `increment` isn't defined.

It's important that we don't waste each others time. Please post a complete and minimal program that can be used to reproduce the error. Please verify that the program posted can be used to reproduce the error. I need the exact code.

Janaka-Steph, 2020-02-22:
You are right, I am sorry. I have updated the example and tested it. It starts at 500 and goes to 2500, then it is supposed to turn it off but does not.

fivdi, 2020-02-22:
The program calls servoWrite with 500 (most anti-clockwise) to 2500 (most clockwise) in steps of 100, then calls servoWrite(0) and then terminates. As the program terminates, it can't be the program that is holding the servo in position. The program can't generate pulses to hold the servo in position after it terminates.

What I think is that you haven't calibrated the servo to find out what the min and max values are that can be passed to servoWrite. I would imagine that passing the value 2500 to servoWrite has forced the servo so far clockwise that it's hard to move the shaft manually and that a lot of pressure needs to be applied to achieve this.

Change this code
```js
const closePosition = 500
const openPosition = 2500
```
to this
```js
const closePosition = 1000
const openPosition = 2000
```
to see if it's easier to move the servo shaft after the program terminates.

The min and max values that can be passed to servoWrite are a different for each servo so calibration is needed for each servo.

Janaka-Steph, 2020-02-24:
Hi! I tried your suggestion, moving between 1000-2000, even 1300-1700, but I have the same issue. I will buy a different more expensive servo, maybe those cheap Tower Pro SG90 are defectives.
Maybe having external power supply would change something?
Also my machine have five servos connected to a RPI 3B, I don't know if it is actually a real issue if all five hold position, maybe the board can handle it.

Janaka-Steph, 2020-02-25:
Something I didn't mention but may be relevant is that even when I stop my node app with Ctrl+C the servo is not released. Switching off the Pi with `sudo shutdown now` don't release the servo either. I have to unplug the power supply. This is really weird.

fivdi, 2020-02-25:
Shutting down the Raspberry Pi doesn't remove the power from the power pins on the GPIO header.

Janaka-Steph, 2020-02-26:
I solved my problem. The issue was the servos Tower Pro SG90 I used. I tried with a Tower Pro MG90S and it works like a charm.  Thank you for your help!

closed 2020-02-26 by Janaka-Steph