fivdi/pigpio #106: How to run pigpio with root privilege?

nicolasherrbach, 2020-04-25:
Hello,

**Problem:**
I'm trying to write a plugin for Homebridge using the pigpio library to send waveforms to a 433 MHz emitter plugged on the Pi's GPIO.

I understand the requirement to use root privileges to run pigpio. And my script works correctly if I run it with sudo.

However, Homebridge runs as a service (hb-service) on the Pi, and I don't have the option to put sudo, so I get the error:

```
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+
```

**What I've tried so far (without success):**
* Allow user pi to run node with the rights of root: `sudo chmod u+s /usr/bin/node`. I still get the error

* Run hb-service as root: https://github.com/oznu/homebridge-config-ui-x/issues/626. Also without success

Is it because the script is ran from a service? Do yo have any idea what I could try?

**Environment:**

* Raspberry Pi 4 B
* Raspbian Buster Lite fully upgraded
* Node.js Version: v12.16.2
* NPM Version: 6.14.4

Node.js installed using:
```
curl -sL https://deb.nodesource.com/setup_12.x | sudo bash -
sudo apt install nodejs
```

fivdi, 2020-04-26:
The source of the error message is [this block of code](https://github.com/joan2937/pigpio/blob/v75/pigpio.c#L7287-L7296) in the pigpio C library.

[This comment](https://github.com/oznu/homebridge-config-ui-x/issues/626#issuecomment-616445189) suggests adding the following to `/etc/systemd/system/homebridge.service`:

```
[Service]
Type=simple
User=root
```
Does it work if no user is specified as the default user for services is `root`?
```
[Service]
Type=simple
```
If not, does it work if both `User` and `Group` are set to `root`?
```
[Service]
Type=simple
User=root
Group=root
```

nicolasherrbach, 2020-04-26:
Thank you for your answer! Unfortunately, none of those options work.
When I look at different processes, here's what I get:

* `$ sudo homebridge` => **no pigpio error**
I get the following homebridge process:
```
$ ps -u -p 2067
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root      2067 19.2  2.1 106608 41976 pts/2    SLl+ 15:44   0:01 homebridge
```

*  `sudo hb-service start` => **pigpio error**
I get the following processes (homebridge process is started by `child_process.fork()` in hb-service js code):
```
$ ps -u -p 1843
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root      1843  0.5  3.3 155148 66880 ?        Ssl  15:20   0:08 hb-service
```
```
ps -u -p 1857
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root      1857  0.1  2.0 129844 40720 ?        Sl   15:20   0:03 homebridge
```

We see that all processes are owned by root, but in some cases it doesn't trigger the pigpio error and in other cases it does. My knowledge of linux is quite limited and I'm not sure how this can happen. However the STAT values of those processes are different. Could it be the cause of the problem?

Here's the meaning of those STAT values (from `man ps`):
```
S    interruptible sleep (waiting for an event to complete)
L    has pages locked into memory (for real-time and custom IO)
s    is a session leader
l    is multi-threaded (using CLONE_THREAD, like NPTL pthreads do)
+    is in the foreground process group
```

fivdi, 2020-04-26:
I'm afraid I'm not familiar enough with Homebridge or using pigpio in services to be able to figure out how to resolve the issue

nicolasherrbach, 2020-05-01:
I've got a solution for this on the [Raspberry Pi stackexchange](https://raspberrypi.stackexchange.com/questions/111884/pigpio-wont-initialise-when-used-by-a-script-started-by-a-service-even-with-roo).

The problem was not linked with pigpio, but with the service file `/etc/systemd/system/homebridge.service` where the default settings from the Homebridge installation were preventing somehow to use root privilege.

A working service file is:
```
[Unit]
Description=Homebridge
Wants=network-online.target
After=syslog.target network-online.target

[Service]
WorkingDirectory=/var/lib/homebridge
EnvironmentFile=/etc/default/homebridge
ExecStartPre=/bin/run-parts /etc/hb-service/homebridge/prestart.d
ExecStartPre=/usr/lib/node_modules/homebridge-config-ui-x/dist/bin/hb-service.js before-start $HOMEBRIDGE_OPTS
ExecStart=/usr/lib/node_modules/homebridge-config-ui-x/dist/bin/hb-service.js run $HOMEBRIDGE_OPTS
KillMode=process

[Install]
WantedBy=multi-user.target
```

Thank you again for your time and effort on this subject!

closed 2020-05-01 by nicolasherrbach

fivdi, 2020-05-01:
@nicolasherrbach good to hear that it's working now and thank you for providing the feedback,