fivdi/pigpio #107: Running on non-pi

vypxl, 2020-05-01:
Can there be any way to install and run this library on a development machine, which does not have pigpio installed? I'm thinking of some callback to be called if the library isn't detected so that the developer can automatically select the [mock library](https://github.com/deepsyx/pigpio-mock) for example. Especially because there is no easy way to exclude a package with `npm install`.

fivdi, 2020-05-01:
Does it work if `pigpio` is defined as an optional dependency rather than a dependency and `pigpio-mock` is defined as a development dependency with something like this in package.json:

```
  "dependencies": {
    // Don't add pogpio as a dependency, add it as an optional dependency
  },

  "optionalDependencies": {
    "pigpio": "^3.1.1"
  },

  "devDependencies": {
    "pigpio-mock": "^0.0.1",
  },

```
Note that there will most likely be compile errors when an attempt is made to install the optional dependency `pigpio` on the development machine. However, these errors can be ignored and your application should install successfully.

fivdi, 2020-05-01:
The [installation section of the pigpio-mock readme](https://github.com/deepsyx/pigpio-mock#installation) shows how to distinguish between production and development environments at runtime.

vypxl, 2020-05-01:
That's true, it only bugs me that I have to accept the compile error every time I install my packages.

fivdi, 2020-05-01:
Yes, compile errors can be annoying.

If your development machine is a mac or PC with the appropriate C/C++ compiler installed there shouldn't be any compile errors as the code is [only compiled on Linux machines](https://github.com/fivdi/pigpio/blob/v3.1.1/binding.gyp#L5). As far as I remember the compiler still needs to be installed though.

vypxl, 2020-05-01:
My workaround on my Linux dev machine is, that I have pigpio installed there, only that when I start it, it obviously gives me an error when I try to start it. It just feels like a hack.

My suggestion is, that this package could detect if it is being installed on a pi or not, and if not, does not throw an error if the build fails.

fivdi, 2020-05-01:
> My suggestion is, that this package could detect if it is being installed on a pi or not, and if not, does not throw an error if the build fails.

This isn't possible. If the build fails the package isn't installed so it can't "not throw an error if the build fails". The error comes from Node.js because `require('pigpio')` fails as the package isn't installed.

vypxl, 2020-05-01:
Ok, maybe I misunderstood that part.

But what I think is possible, would be to add a condition to the binding.gyp file which checks `/proc/cpuinfo` and if it contains the string 'Raspberry Pi', like `grep "Raspberry Pi" /proc/cpuinfo`. Then one could use a different dummy source file without the reference to the pigpio library.

I understand if this is out of scope though.

fivdi, 2020-05-01:
Try `npm install fivdi/pigpio#compile` to see if it works for you.

vypxl, 2020-05-01:
Works like a charm, this is exactly what I hoped for!

I would only change `pigpio.js#L6` to 
```js 
const pigpio = (() => {
----------------^^-----```

fivdi, 2020-05-01:
Why would you change that line?

A side effect of this enhancement is that pigpio can now have a travis build :smile:

vypxl, 2020-05-01:
I would change it, because using `()` means 'no function parameters' and a `_` is an unused function parameter.

Nice thing with Travis :grin:

fivdi, 2020-05-01:
Ok. Thanks for the tip.

I'll release a new version of pigpio tomorrow.

vypxl, 2020-05-01:
Thanks for your quick help :)

closed 2020-05-02 by fivdi

fivdi, 2020-05-02:
pigpio@3.2.0 which includes this enhancement has now been published on npm.

PR #108 (2020-05-02): ensure that pigpio installs without errors on all linux systems 
Fixes #107