fivdi/pigpio #109: Failed compiling / installing 3.2.0

highstrike, 2020-05-02:
Sup?

So I tried building a docker image for my raspberry pi 4 but I can't install the newest version which is 3.2.0. However it works just fine with 3.1.1.

This is my [Dockerfile](https://github.com/highstrike/raspberry-ready/blob/69a616eae8a64ccb6b3cdfc280ffc35cadf05720/Dockerfile) and I just built the image ignoring the last 2 lines so that I can go in it with exec to check things out manually.

So I just have a `package.json` file and the `start.js` script in the `/app` directory, all of the dependencies are are installed correctly and you can see the versions here:
[image]

After running `npm install` with `pigpio:3.2.0` as a dependency I get the following error

```
/app # cat /root/.npm/_logs/2020-05-02T15_29_41_318Z-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'install' ]
2 info using npm@6.14.4
3 info using node@v12.16.2
4 verbose npm-session 94f4536c497ac96a
5 silly install runPreinstallTopLevelLifecycles
6 silly preinstall raspberry-ready@1.0.3
7 info lifecycle raspberry-ready@1.0.3~preinstall: raspberry-ready@1.0.3
8 silly install loadCurrentTree
9 silly install readLocalPackageData
10 timing stage:loadCurrentTree Completed in 21ms
11 silly install loadIdealTree
12 silly install cloneCurrentTreeToIdealTree
13 timing stage:loadIdealTree:cloneCurrentTree Completed in 1ms
14 silly install loadShrinkwrap
15 timing stage:loadIdealTree:loadShrinkwrap Completed in 2ms
16 silly install loadAllDepsIntoIdealTree
17 http fetch GET 304 https://registry.npmjs.org/pigpio 275ms (from cache)
18 silly pacote range manifest for pigpio@^3.1.1 fetched in 296ms
19 http fetch GET 304 https://registry.npmjs.org/bindings 67ms (from cache)
20 silly pacote range manifest for bindings@^1.5.0 fetched in 73ms
21 http fetch GET 304 https://registry.npmjs.org/nan 129ms (from cache)
22 silly pacote range manifest for nan@^2.14.1 fetched in 138ms
23 http fetch GET 304 https://registry.npmjs.org/file-uri-to-path 82ms (from cache)
24 silly pacote version manifest for file-uri-to-path@1.0.0 fetched in 87ms
25 silly pacote range manifest for pigpio@^3.1.1 fetched in 4ms
26 silly resolveWithNewModule pigpio@3.2.0 checking installable status
27 silly pacote range manifest for bindings@^1.5.0 fetched in 6ms
28 silly resolveWithNewModule bindings@1.5.0 checking installable status
29 silly pacote range manifest for nan@^2.14.1 fetched in 8ms
30 silly resolveWithNewModule nan@2.14.1 checking installable status
31 silly pacote version manifest for file-uri-to-path@1.0.0 fetched in 2ms
32 silly resolveWithNewModule file-uri-to-path@1.0.0 checking installable status
33 timing stage:loadIdealTree:loadAllDepsIntoIdealTree Completed in 506ms
34 timing stage:loadIdealTree Completed in 515ms
35 silly currentTree raspberry-ready@1.0.3
36 silly idealTree raspberry-ready@1.0.3
36 silly idealTree +-- bindings@1.5.0
36 silly idealTree +-- file-uri-to-path@1.0.0
36 silly idealTree +-- nan@2.14.1
36 silly idealTree `-- pigpio@3.2.0
37 silly install generateActionsToTake
38 timing stage:generateActionsToTake Completed in 20ms
39 silly diffTrees action count 4
40 silly diffTrees add file-uri-to-path@1.0.0
41 silly diffTrees add bindings@1.5.0
42 silly diffTrees add nan@2.14.1
43 silly diffTrees add pigpio@3.2.0
44 silly decomposeActions action count 32
45 silly decomposeActions fetch file-uri-to-path@1.0.0
46 silly decomposeActions extract file-uri-to-path@1.0.0
47 silly decomposeActions preinstall file-uri-to-path@1.0.0
48 silly decomposeActions build file-uri-to-path@1.0.0
49 silly decomposeActions install file-uri-to-path@1.0.0
50 silly decomposeActions postinstall file-uri-to-path@1.0.0
51 silly decomposeActions finalize file-uri-to-path@1.0.0
52 silly decomposeActions refresh-package-json file-uri-to-path@1.0.0
[... 116 lines cut ...]
159 verbose stack     at maybeClose (internal/child_process.js:1021:16)
159 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
160 verbose pkgid pigpio@3.2.0
161 verbose cwd /app
162 verbose Linux 4.19.97-v7l+
163 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "install"
164 verbose node v12.16.2
165 verbose npm  v6.14.4
166 error code ELIFECYCLE
167 error errno 1
168 error pigpio@3.2.0 install: `node-gyp rebuild`
168 error Exit status 1
169 error Failed at the pigpio@3.2.0 install script.
169 error This is probably not a problem with npm. There is likely additional logging output above.
170 verbose exit [ 1, true ]
```

But switching to `pigpio:3.1.1` and it works just fine:
[image]

I'm also running the latest raspbian buster with the following kernel
[image]

And this is my docker version
[image]

highstrike, 2020-05-02:
Forgot to add, running `npm install` with `3.2.0` directly on my raspberry pi, it works, it just doesn't work in docker while `3.1.1` works fine there.

fivdi, 2020-05-02:
Sorry about that.

What's new in pigpio@3.2.0 is [this line in binding.gyp](https://github.com/fivdi/pigpio/blob/v3.2.0/binding.gyp#L7):
```
          '"<!(ldconfig -p | grep libpigpio.so | wc -l)" != "0"', {
```
It basically runs the command `ldconfig -p | grep libpigpio.so | wc -l` to see if the pigpic C library has been installed. The pigpio Node.js module depends on the pigpio C libarry.

Is it possible that the Docker image has problems running the command `ldconfig -p | grep libpigpio.so | wc -l`?

There appears to be something special about `ldconfig` in the Dockerfile. For example see [this line](https://github.com/highstrike/raspberry-ready/blob/v1.0.4/Dockerfile#L12):
```
    sed -i 's/ldconfig/ldconfig \/usr\/local/g' Makefile && \
```

highstrike, 2020-05-02:
So the build instruction you mentioned changes the `Makefile` file by adding `/usr/local` to `ldconfig` as you can see below
[image]

Otherwise the `make install` command fails.

Indeed docker (on alpine linux) has an issue with the `-p` argument
[image]

Hmm

fivdi, 2020-05-02:
What the docker (on alpine Linux) output for the following command?
```
strings /etc/ld.so.cache | grep libpigpio.so
```

highstrike, 2020-05-02:
[image]

apparently alpine doesn't have such a file

fivdi, 2020-05-02:
>apparently alpine doesn't have such a file

Indeed. It doesn't appear to have very much.

What does docker (on alpine Linux) output for the following command?
```
which pigpiod
```

highstrike, 2020-05-02:
Yeah, alpine is a very light weight linux distro, very popular with docker.

It found the pigpiod in `/usr/local/bin/pigpiod`

[image]

fivdi, 2020-05-02:
Ok, that's good. It should be possible to fix the problem.

highstrike, 2020-05-02:
Maybe this can also be of assistance

[image]

fivdi, 2020-05-02:
`find` would work but it will be very slow on a Raspberry Pi 0. Also, I don't want to use `sudo` and without `sudo` there will be a ton of error messages. `which pigpiod` is good.

81d205fb3a812c56b9423ce9c5369bd14522cae2 should fix the problem.

Please give it a try with pigpio@3.2.1 which has just been published on npm.

highstrike, 2020-05-02:
Great success! You fixed it in `3.2.1`

Thank you! 🙇 

```
pi@raspberrypi:~/github/raspberry-ready $ docker build -t highstrike/raspberry-ready .
Sending build context to Docker daemon  120.3kB
Step 1/7 : FROM arm32v7/node:12.16.2-alpine
12.16.2-alpine: Pulling from arm32v7/node
3cfb62949d9d: Pull complete
f6fef232cef8: Pull complete
b4f416b96422: Pull complete
da5a829b9572: Pull complete
Digest: sha256:63377c2c43582a08c3db7235d051886cb313007b1050b8243f3455ba68e29012
Status: Downloaded newer image for arm32v7/node:12.16.2-alpine
 ---> 4ded18f99365
Step 2/7 : RUN apk --update --no-cache add --virtual dependencies curl build-base python &&     cd /tmp &&     curl -s https://api.github.com/repos/joan2937/pigpio/releases/latest     | grep 'tarball_url.*'     | cut -d : -f 2,3     | tr -d \",     | xargs -n 1 curl -sSL     | tar -xz --strip-components=1 &&     make &&     sed -i 's/ldconfig/ldconfig \/usr\/local/g' Makefile &&     make install
 ---> Running in e4b46737c6bb
fetch http://dl-cdn.alpinelinux.org/alpine/v3.11/main/armv7/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.11/community/armv7/APKINDEX.tar.gz
(1/30) Installing ca-certificates (20191127-r1)
(2/30) Installing nghttp2-libs (1.40.0-r0)
(3/30) Installing libcurl (7.67.0-r0)
(4/30) Installing curl (7.67.0-r0)
(5/30) Installing binutils (2.33.1-r0)
(6/30) Installing libmagic (5.37-r1)
(7/30) Installing file (5.37-r1)
(8/30) Installing gmp (6.1.2-r1)
(9/30) Installing isl (0.18-r0)
(10/30) Installing libgomp (9.2.0-r4)
(11/30) Installing libatomic (9.2.0-r4)
(12/30) Installing mpfr4 (4.0.2-r1)
(13/30) Installing mpc1 (1.1.0-r1)
(14/30) Installing gcc (9.2.0-r4)
(15/30) Installing musl-dev (1.1.24-r2)
(16/30) Installing libc-dev (0.7.2-r0)
(17/30) Installing g++ (9.2.0-r4)
(18/30) Installing make (4.2.1-r2)
(19/30) Installing fortify-headers (1.1-r0)
(20/30) Installing build-base (0.5-r1)
(21/30) Installing libbz2 (1.0.8-r1)
(22/30) Installing expat (2.2.9-r1)
(23/30) Installing libffi (3.2.1-r6)
(24/30) Installing gdbm (1.13-r1)
(25/30) Installing ncurses-terminfo-base (6.1_p20200118-r4)
(26/30) Installing ncurses-libs (6.1_p20200118-r4)
(27/30) Installing readline (8.0.1-r0)
(28/30) Installing sqlite-libs (3.30.1-r1)
(29/30) Installing python2 (2.7.16-r3)
(30/30) Installing dependencies (20200502.200551)
Executing busybox-1.31.1-r9.trigger
Executing ca-certificates-20191127-r1.trigger
OK: 160 MiB in 46 packages
gcc -O3 -Wall -pthread -fpic -c -o pigpio.o pigpio.c
gcc -O3 -Wall -pthread -fpic -c -o command.o command.c
gcc -shared -pthread -Wl,-soname,libpigpio.so.1 -o libpigpio.so.1 pigpio.o command.o
ln -fs libpigpio.so.1 libpigpio.so
strip --strip-unneeded libpigpio.so
size     libpigpio.so
   text    data     bss     dec     hex filename
 222727    6116  605548  834391   cbb57 libpigpio.so
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if.o pigpiod_if.c
gcc -shared -pthread -Wl,-soname,libpigpiod_if.so.1 -o libpigpiod_if.so.1 pigpiod_if.o command.o
ln -fs libpigpiod_if.so.1 libpigpiod_if.so
strip --strip-unneeded libpigpiod_if.so
size     libpigpiod_if.so
   text    data     bss     dec     hex filename
  44147    5036   49272   98455   18097 libpigpiod_if.so
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if2.o pigpiod_if2.c
gcc -shared -pthread -Wl,-soname,libpigpiod_if2.so.1 -o libpigpiod_if2.so.1 pigpiod_if2.o command.o
[... 103 lines cut ...]
(19/30) Purging mpc1 (1.1.0-r1)
(20/30) Purging mpfr4 (4.0.2-r1)
(21/30) Purging gmp (6.1.2-r1)
(22/30) Purging python2 (2.7.16-r3)
(23/30) Purging libbz2 (1.0.8-r1)
(24/30) Purging expat (2.2.9-r1)
(25/30) Purging libffi (3.2.1-r6)
(26/30) Purging gdbm (1.13-r1)
(27/30) Purging readline (8.0.1-r0)
(28/30) Purging ncurses-libs (6.1_p20200118-r4)
(29/30) Purging ncurses-terminfo-base (6.1_p20200118-r4)
(30/30) Purging sqlite-libs (3.30.1-r1)
Executing busybox-1.31.1-r9.trigger
OK: 5 MiB in 16 packages
Removing intermediate container 0c88218a570b
 ---> ab3717c5309f
Step 6/7 : COPY start.js /app
 ---> 7a3e8827ed90
Step 7/7 : CMD ["npm", "start"]
 ---> Running in 0c6675095ec2
Removing intermediate container 0c6675095ec2
 ---> 7401d98f25b4
Successfully built 7401d98f25b4
Successfully tagged highstrike/raspberry-ready:latest
```

closed 2020-05-02 by highstrike

fivdi, 2020-05-02:
Excellent, thanks for the feedback.

poegun, 2024-09-10:
pigpiod.service: Failed at step EXEC spawning /usr/local/bin/pigpiod: No such file or directory

I've been dealing with this problem for 10 days, no matter what I tried, I bought rpi5 for hobby purposes, what nonsense!