fivdi/pigpio #112: Error: Module did not self-register

DoganM95, 2020-05-10:
OS: Raspbian 10 (Buster, Lite)
Model Raspberry Pi 3B
Node: 14.2.0
pigpio: 3.2.2

I am getting the following error when trying to run my webserver.js which contains this line: `let pigpioModule = require("pigpio");` :

```
+---------------------------------------------------+
| Warning: The pigpio C library can't be loaded on  |
| this machine and any attempt to use it will fail. |
+---------------------------------------------------+

Invoking require('bindings')('pigpio.node') resulted in the follwoing error:

Error: Module did not self-register: '/home/pi/project/RaspberryPi/Software/node_modules/pigpio/build/Release/pigpio.node'.
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1226:18)
    at Module.load (internal/modules/cjs/loader.js:1040:32)
    at Function.Module._load (internal/modules/cjs/loader.js:929:14)
    at Module.require (internal/modules/cjs/loader.js:1080:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at bindings (/home/pi/project/RaspberryPi/Software/node_modules/bindings/bindings.js:112:48)
    at /home/pi/project/RaspberryPi/Software/node_modules/pigpio/pigpio.js:8:31
    at Object.<anonymous> (/home/pi/project/RaspberryPi/Software/node_modules/pigpio/pigpio.js:33:3)
    at Module._compile (internal/modules/cjs/loader.js:1176:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1196:10)

-------------------------------------------------------------------------------

/home/pi/project/RaspberryPi/Software/node_modules/pigpio/pigpio.js:446
  pigpio.gpioCfgClock(+microseconds, +peripheral);
         ^

TypeError: pigpio.gpioCfgClock is not a function
    at Object.module.exports.configureClock (/home/pi/project/RaspberryPi/Software/node_modules/pigpio/pigpio.js:446:10)
    at Object.<anonymous> (/home/pi/project/RaspberryPi/Software/src/main/javascript/webserver.js:19:14)
    at Module._compile (internal/modules/cjs/loader.js:1176:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1196:10)
    at Module.load (internal/modules/cjs/loader.js:1040:32)
    at Function.Module._load (internal/modules/cjs/loader.js:929:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:71:12)
    at internal/main/run_main_module.js:17:47
```

after a clean installation on a raspberry pi. any suggestions?

`node --version `
v14.2.0
`sudo node --version`
v14.2.0

fivdi, 2020-05-11:
Please post the complete output of the following commands when run on the Raspberry Pi:
```
which pigpiod
pigpiod -v
```

nidaldh, 2020-05-11:
i have the same problem.

fivdi, 2020-05-11:
@nidaldh Please post the requested information.

DoganM95, 2020-05-11:
`which pigpiod` gave me no output and 
`pigpiod -v` said pigpiod was not installed.

I installed it using
`sudo apt install pigpiod` and rebooted. 

Now the outputs are:
`$ which pigpiod`
**/usr/bin/pigpiod**
`$ pigpiod -v`
**71**

Tried starting server.js again, gave me the following error (same to still be the same):

```
+---------------------------------------------------+
| Warning: The pigpio C library can't be loaded on  |
| this machine and any attempt to use it will fail. |
+---------------------------------------------------+

Invoking require('bindings')('pigpio.node') resulted in the follwoing error:

Error: Module did not self-register: '/home/pi/project/RaspberryPi/Software/node_modules/pigpio/build/Release/pigpio.node'.
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1226:18)
    at Module.load (internal/modules/cjs/loader.js:1040:32)
    at Function.Module._load (internal/modules/cjs/loader.js:929:14)
    at Module.require (internal/modules/cjs/loader.js:1080:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at bindings (/home/pi/project/RaspberryPi/Software/node_modules/bindings/bindings.js:112:48)
    at /home/pi/project/RaspberryPi/Software/node_modules/pigpio/pigpio.js:8:31
    at Object.<anonymous> (/home/pi/project/RaspberryPi/Software/node_modules/pigpio/pigpio.js:33:3)
    at Module._compile (internal/modules/cjs/loader.js:1176:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1196:10)

-------------------------------------------------------------------------------

/home/pi/project/RaspberryPi/Software/node_modules/pigpio/pigpio.js:446
  pigpio.gpioCfgClock(+microseconds, +peripheral);
         ^

TypeError: pigpio.gpioCfgClock is not a function
    at Object.module.exports.configureClock (/home/pi/project/RaspberryPi/Software/node_modules/pigpio/pigpio.js:446:10)
    at Object.<anonymous> (/home/pi/project/RaspberryPi/Software/src/main/javascript/webserver.js:19:14)
    at Module._compile (internal/modules/cjs/loader.js:1176:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1196:10)
    at Module.load (internal/modules/cjs/loader.js:1040:32)
    at Function.Module._load (internal/modules/cjs/loader.js:929:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:71:12)
    at internal/main/run_main_module.js:17:47
```

fivdi, 2020-05-11:
@DoganM95 Please reinstall the pipgio Node.js module and it should work correctly. Let us know it works as expected.

The warning messages needs to be improved as it doesn't say enough.

DoganM95, 2020-05-11:
uninstalled pigpio successfully using
`npm uninstall pigpio`

tried reinstalling using 
`npm install pigpio`

which gave me the following error:

```
> pigpio@3.2.2 install /home/pi/project/RaspberryPi/Software/node_modules/pigpio
> node-gyp rebuild

make: Entering directory '/home/pi/project/RaspberryPi/Software/node_modules/pigpio/build'
  CXX(target) Release/obj.target/pigpio/src/pigpio.o
../src/pigpio.cc:2:10: fatal error: pigpio.h: No such file or directory
 #include <pigpio.h>
          ^~~~~~~~~~
compilation terminated.
make: *** [pigpio.target.mk:113: Release/obj.target/pigpio/src/pigpio.o] Error 1
make: Leaving directory '/home/pi/project/RaspberryPi/Software/node_modules/pigpio/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:276:12)
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/project/RaspberryPi/Software/node_modules/pigpio
gyp ERR! node -v v14.2.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm WARN rc-car@1.0.0 No description

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! pigpio@3.2.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the pigpio@3.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-05-11T15_35_49_769Z-debug.log
```

tried again using 
`sudo npm install pigpio`
ang got a different error message

```
> pigpio@3.2.2 install /home/pi/project/RaspberryPi/Software/node_modules/pigpio
> node-gyp rebuild

gyp WARN EACCES current user ("pi") does not have permission to access the dev dir "/root/.cache/node-gyp/14.2.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/project/RaspberryPi/Software/node_modules/pigpio/.node-gyp"
gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/pi/project/RaspberryPi/Software/node_modules/pigpio/.node-gyp'
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/project/RaspberryPi/Software/node_modules/pigpio
gyp ERR! node -v v14.2.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm WARN rc-car@1.0.0 No description

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! pigpio@3.2.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the pigpio@3.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-05-11T15_42_19_376Z-debug.log
```

installing other packages works fine, e.g. :
`npm install express` gave me the folllowing

```
...
+ express@4.17.1
added 52 packages from 34 contributors and audited 220 packages in 11.063s
...
```

fivdi, 2020-05-11:
@DoganM95 I think we have it now. Please follow both step 1 and step 2 of the [installation instructions](https://github.com/fivdi/pigpio#installation) exactly as specified in the README. The correct command for installing the pigpio C library is `sudo apt-get install pigpio` and not `sudo apt install pigpiod` as indicated [this comment](https://github.com/fivdi/pigpio/issues/112#issuecomment-626714082). Let us know if it then works as expected.

DoganM95, 2020-05-11:
Yes, it works now, `sudo apt-get install pigpio` solved the problem.

Thank you very much for your help!

closed 2020-05-11 by DoganM95

fivdi, 2020-05-11:
You're welcome. Good to hear that it's working now.