fivdi/pigpio #113: Take too much time to release the GPIO

CorentinDy, 2020-06-08:
Hi,
I've develop an express application which include **pigpio** in order to set on/off a 360 continuous servo motor.

The code is "working" but sometimes it take too much time to release the GPIO..

```
POST /feed 200 39.094 ms - 14
POST /feed 200 35.693 ms - 14
POST /feed 200 54.325 ms - 14
POST /feed 200 47.295 ms - 14
POST /feed 200 39.242 ms - 14
POST /feed 200 36.266 ms - 14
POST /feed 200 39.847 ms - 14
POST /feed 200 1444.039 ms - 14
```

The last one is way too long and I can't explain it.
I've tried many rpi board (zero/2/3/4).

- Zero -> crash and motor keep spinning until I stop it manually
- 2/3/4 -> take time to stop the motor

I'm sharing with you my code here : [Github](https://github.com/CorentinDy/feed-the-kakashi)
I hope it's clean enough to be understand by anyone .

The file related to turn on and off the motor is located here [feeder.js](https://github.com/CorentinDy/feed-the-kakashi/blob/master/utils/feeder.js)
(yes the timeout in useful, just to turn on for a little amount of time)

The servo motor is [this one](http://www.domanrchobby.com/content/?150.html). DM-S0306D

I hope someone can find a solution and I really hope it's a dummy bug made by myself !

fivdi, 2020-06-08:
Here's the code from the [feed function](https://github.com/CorentinDy/feed-the-kakashi/blob/master/utils/feeder.js#L5-L15) at the moment:
```js
function feed() {
    const motor = new Gpio(13, { mode: Gpio.OUTPUT });
    // Init the motor GPIO
    motor.servoWrite(1500);

    motor.servoWrite(2500);
    setTimeout(function () {
        motor.servoWrite(1500);
    }, 200);
    db.store();
}
```
To see how long it takes the pigpio code to execute modify the feed function to the following:
```js
function feed() {
    let time = process.hrtime();

    const motor = new Gpio(13, { mode: Gpio.OUTPUT });
    // Init the motor GPIO
    // motor.servoWrite(1500); <<< === REMOVE THIS LINE === >>>

    motor.servoWrite(2500);

    time = process.hrtime(time);
    console.log("pigpio1 " + ((time[0] + time[1] / 1E9) * 1000) + " ms"); // <<<=== Should be far less that 1 ms === >>>

    time = process.hrtime();
    setTimeout(function () {
        motor.servoWrite(1500);
        console.log("pigpio2 " + ((time[0] + time[1] / 1E9) * 1000) + " ms"); // <<<=== Should be approximately 200 ms === >>>
    }, 200);
    db.store();
}
```
The modified code will log two times to the console. The first ("pigpio1") should be far less than 1 millisecond. The second ("pigpio2") should be approximately 200 milliseconds.

The long delays are likely to be related to your network infrastructure or other parts of the code not related to pigpio. For example, the last line of function feed is:
```js
    db.store();
```
How long does it take db.store() to execute? You could use the process.hrtime to determine the answer.

I don't think this issues is related to pigpio.

CorentinDy, 2020-06-09:
Thanks for the really well explain answer !
So here are the logs with the timer

```
pigpio1 366.554345 ms
POST /feed 200 1864.326 ms - 14
pigpio2 1490.638253 ms
pigpio1 0.217839 ms
POST /feed 200 1495.088 ms - 14
pigpio2 1497.608777 ms
pigpio1 0.125275 ms
POST /feed 200 1468.990 ms - 14
pigpio2 1471.241587 ms
pigpio1 0.12501299999999999 ms
POST /feed 200 68.947 ms - 14
pigpio2 200.480126 ms
pigpio1 0.130169 ms
POST /feed 200 72.081 ms - 14
pigpio2 200.557851 ms
pigpio1 0.148452 ms
POST /feed 200 72.130 ms - 14
pigpio2 200.598531 ms
pigpio1 0.191477 ms
POST /feed 200 145.618 ms - 14
pigpio2 200.226761 ms
```

The `db.store();` is a new feature, the bug was here before this function, but to test your code ,I've commented it.
I've still tested with the store function , the average response time is about 32ms

fivdi, 2020-06-09:
I downloaded the application and gave it a try. As can be seen in the output below it's fairly easy to reproduce the problem:

```
GET / 200 97.547 ms - 102092
GET /bts-css/bootstrap.min.css 304 6.486 ms - -
GET /fa-css/all.min.css 304 5.347 ms - -
GET /flatpickr/flatpickr.min.css 304 4.732 ms - -
GET /notif/css/iziToast.css 304 3.103 ms - -
GET /stylesheets/style.css 304 0.965 ms - -
GET /stylesheets/mini-event-calendar.min.css 304 1.002 ms - -
GET /bts-js/bootstrap.min.js 304 1.266 ms - -
GET /javascript/mini-event-calendar.min.js 304 1.555 ms - -
GET /fa-js/all.min.js 304 8.474 ms - -
GET /notif/js/iziToast.js 304 8.334 ms - -
POST /feed 200 351.689 ms - 14
POST /feed 200 55.782 ms - 14
POST /feed 200 61.585 ms - 14
POST /feed 200 55.054 ms - 14
POST /feed 200 51.560 ms - 14
POST /feed 200 61.663 ms - 14
POST /feed 200 63.991 ms - 14
POST /feed 200 62.628 ms - 14
POST /feed 200 61.394 ms - 14
POST /feed 200 56.958 ms - 14
POST /feed 200 63.983 ms - 14
POST /feed 200 60.824 ms - 14
POST /feed 200 45.842 ms - 14
POST /feed 200 64.436 ms - 14
POST /feed 200 70.622 ms - 14
POST /feed 200 55.300 ms - 14
POST /feed 200 61.897 ms - 14
POST /feed 200 60.259 ms - 14
POST /feed 200 65.721 ms - 14
POST /feed 200 54.637 ms - 14
POST /feed 200 449.788 ms - 14
POST /feed 200 929.430 ms - 14
POST /feed 200 64.337 ms - 14
POST /feed 200 684.065 ms - 14
POST /feed 200 58.806 ms - 14
POST /feed 200 59.997 ms - 14
POST /feed 200 60.266 ms - 14
POST /feed 200 689.711 ms - 14
POST /feed 200 77.285 ms - 14
POST /feed 200 69.531 ms - 14
POST /feed 200 67.784 ms - 14
POST /feed 200 472.335 ms - 14
POST /feed 200 59.759 ms - 14
POST /feed 200 63.299 ms - 14
POST /feed 200 56.622 ms - 14
POST /feed 200 58.174 ms - 14
POST /feed 200 49.506 ms - 14
POST /feed 200 73.714 ms - 14
POST /feed 200 56.595 ms - 14
POST /feed 200 53.697 ms - 14
POST /feed 200 62.224 ms - 14
POST /feed 200 61.338 ms - 14
POST /feed 200 57.587 ms - 14
POST /feed 200 52.143 ms - 14
POST /feed 200 56.979 ms - 14
POST /feed 200 64.333 ms - 14
POST /feed 200 60.386 ms - 14
POST /feed 200 60.651 ms - 14
POST /feed 200 54.453 ms - 14
POST /feed 200 61.530 ms - 14
POST /feed 200 60.744 ms - 14
POST /feed 200 55.889 ms - 14
POST /feed 200 48.588 ms - 14
POST /feed 200 66.940 ms - 14
POST /feed 200 65.708 ms - 14
POST /feed 200 53.473 ms - 14
[... 602 lines cut ...]
POST /feed 200 1.239 ms - 14
POST /feed 200 1.241 ms - 14
POST /feed 200 1.020 ms - 14
POST /feed 200 1.243 ms - 14
POST /feed 200 1.294 ms - 14
POST /feed 200 1.320 ms - 14
POST /feed 200 1.111 ms - 14
POST /feed 200 1.027 ms - 14
POST /feed 200 1.310 ms - 14
POST /feed 200 1.284 ms - 14
POST /feed 200 1.022 ms - 14
POST /feed 200 1.286 ms - 14
POST /feed 200 1.637 ms - 14
POST /feed 200 1.046 ms - 14
POST /feed 200 1.293 ms - 14
POST /feed 200 1.236 ms - 14
POST /feed 200 1.235 ms - 14
POST /feed 200 1.281 ms - 14
POST /feed 200 1.285 ms - 14
POST /feed 200 1.238 ms - 14
POST /feed 200 1.285 ms - 14
POST /feed 200 1.246 ms - 14
POST /feed 200 1.301 ms - 14
```
It looks like the issue is related to `db.store()` and not pigpio.

CorentinDy, 2020-06-10:
Damn ! 
Well, your library is too strong to be defeated by anything .
Thanks for taking the time to answer me so well ! 

It was really the `db.store()` function.

closed 2020-06-10 by CorentinDy