fivdi/pigpio #114: Is there a way to run an app using pigpio without root permissions

jrgleason, 2020-06-09:
The answer to this might be quick and no. Since I can't use the daemon, how do I run an app using this lib without root privileges?

fivdi, 2020-06-09:
I'm afraid it's not possible to run an app that uses the pigpio Node.js module without root permissions. The pigpio Node.js module is a wrapper for the [pigpio C library](https://github.com/joan2937/pigpio) and the C library requires root permissions.

closed 2020-06-11 by fivdi

dceejay, 2020-07-09:
I know you've closed this - but would it be possible to optionally build it against the pigpiod_if2 interface which then goes via the gpiod daemon but doesn't require root...  Are the interfaces similar enough for it to work albeit with less performance ?

fivdi, 2020-07-10:
The pigpiod_if2 interface isn't a drop in replacement for the pigpio C interface and functions that are used to achieve the same thing, for example, gpio_read from the pigpiod_if2 interface and gpioRead from thepigpio C interfaces are named differently and have different parameter lists.

In addition, the pigpio C interface is non-blocking and can be called synchronously but the pigpiod_if2 interface is blocking and should be called asynchronously.

In other words, achieving what you would like to achieve would be a significant amount of work.

dceejay, 2020-07-10:
Yup - understood...  what a pain... never mind.
Thanks

dmccue128, 2022-09-25:
Why is it necessary for pigpio to run as root? I see many statements that the "c library implementation requires root" but no explanation.  It is not necessary to run as root to access gpio, so what is it about the pigpio implementation that requires it to be run as root?  It is a security deal-breaker. Is there no way to fix this?
Can the daemon be run as root and the applications run as normal non-root users?

fivdi, 2022-09-26:
> Why is it necessary for pigpio to run as root? I see many statements that the "c library implementation requires root" but no explanation. It is not necessary to run as root to access gpio, so what is it about the pigpio implementation that requires it to be run as root?

For an authoritative answer to this question you should ask over at the [pigpio C library repository](https://github.com/joan2937/pigpio). My understanding is that the pigpio C library uses /dev/mem in order to access hardware registers and root privileges are required for accessing /dev/mem. Some of the advanced functionality supported by the pigpio C library can only be provided by accessing /dev/mem.

> Is there no way to fix this?

If all you need to do is turn GPIO pins on and off then perhaps you can use [onoff](https://github.com/fivdi/onoff/) or a similar npm package which doesn't require root privileges.

> Can the daemon be run as root and the applications run as normal non-root users?

The pigpio daemon can be run as root and non-root applications can then communicate with the daemon to access GPIOs. However, this is not supported by this repository, that is, the pigpio npm package.

BraveClient, 2022-12-08:
> For an authoritative answer to this question you should ask over at the [pigpio C library repository](https://github.com/joan2937/pigpio). My understanding is that the pigpio C library uses /dev/mem [...]

Counld is use "**sudo chmod 666 /dev/mem**" to using pigpio without root permissions?

fivdi, 2022-12-08:
> Counld is use "**sudo chmod 666 /dev/mem**" to using pigpio without root permissions?

It didn't the last time I checked. See https://github.com/fivdi/pigpio/issues/2#issuecomment-164150969

dmccue128, 2022-12-08:
OK, thanks.  This will be a highly functional library for prototyping and
hobby projects.
For my network-connected IOT application, it is unusable in production.
Maybe I can find another Provider that does not have this requirement or I
will have to abandon pi4j and go to a different solution altogether.
Thanks,
Dan

fivdi, 2022-12-08:
> OK, thanks. This will be a highly functional library for prototyping and hobby projects. For my network-connected IOT application, it is unusable in production. Maybe I can find another Provider [...]

Yep, requiring root privileges is a limitation. Good luck with the search. Note that this repository has nothing to do with pi4j.

dmccue128, 2022-12-08:
Thanks Brian. I apologize if I posted my comment in the wrong place.

FOSSBOSS, 2024-02-28:
> I know you've closed this - but would it be possible to optionally build it against the pigpiod_if2 interface which then goes via the gpiod daemon but doesn't require root... Are the interfaces similar enough for it to work albeit with less performance ?

you can modify the permissions to the pigpio executable like this:
sudo chown root:root prog
sudo chmod 4755 prog

the executable no longer needs root to run. 
Its still annoying, you have to do this every time you recompile, but you can probably write a script to do it.