fivdi/pigpio #115: can't invoke digitalWrite when passing it to another module

zapphyre, 2020-06-22:
I am passing `digitalWrite` func to the other module in which I want to invoke it with 1/0 value based on some logic: 
```
...
const plumbing = require('./f_inc');
const heater = new Gpio(21, {mode: Gpio.OUTPUT});
const controller = plumbing({heater: heater.digitalWrite, ds18b20});
...
```
when I invoke that method I get :
```
.../node_modules/pigpio/pigpio.js:204
    pigpio.gpioWrite(this.gpio, +level);
                          ^

TypeError: Cannot read property 'gpio' of undefined
    at digitalWrite (/home/alarm/incubo_box/node_modules/pigpio/pigpio.js:204:27)
```
so I try to `bind` this on that method when passing it around:
```
.../node_modules/pigpio/pigpio.js:204
    pigpio.gpioWrite(this.gpio, +level);
           ^

Error: EINVAL, Invalid argument
    at Object.digitalWrite (/home/alarm/incubo_box/node_modules/pigpio/pigpio.js:204:12)
...
errno: 22,
  code: 'EINVAL',
  syscall: 'gpioSetPullUpDown'
```
however when I pass just `heater` to the module and invoke `digitalWrite` in the other module it works. However I want the second module to stay agnostic so I would rather pass in concrete function and invoke it in some generic manner.

what can be wrong? 
thanks

fivdi, 2020-06-23:
It should work with something like this:
```js
const plumbing = require('./f_inc');
const heater = new Gpio(21, {mode: Gpio.OUTPUT});
const heaterDigitalWrite = function(value) {
  this.digitalWrite(value);
}.bind(heater);
const controller = plumbing({heater: heaterDigitalWrite, ds18b20});
```
or this:
```js
const plumbing = require('./f_inc');
const heater = new Gpio(21, {mode: Gpio.OUTPUT});
const controller = plumbing({heater: function(value) {
  this.digitalWrite(value);
}.bind(heater), ds18b20});
```

zapphyre, 2020-06-23:
good shot! that solves it, idkw

fivdi, 2020-06-23:
Good to hear that it's working now.

closed 2020-06-23 by fivdi