fivdi/pigpio #116: Array support

IanW69, 2020-07-14:
I have tried to address the GPIO outputs in an array in order to provide a way of controlling GPIO peripherals.

const outputDevice = [];
outputDevice.push = new Gpio(23, {mode: Gpio.OUTPUT});
outputDevice.push = new Gpio(24, {mode: Gpio.OUTPUT});
outputDevice.push = new Gpio(25, {mode: Gpio.OUTPUT});
outputDevice.push = new Gpio(22, {mode: Gpio.OUTPUT});

and then using...

outputDevice[0].digitalWrite(1);

But I try this approach I get the following error "Property 'digitalWrite' does not exist on type 'never'.ts(2339)"

fivdi, 2020-07-14:
The Array `push()` method is being used incorrectly.

In other words, this:
```js
 outputDevice.push = new Gpio(23, {mode: Gpio.OUTPUT});
```
Should be:
```js
 outputDevice.push(new Gpio(23, {mode: Gpio.OUTPUT}));
```

IanW69, 2020-07-14:
Thanks for the update - Unfortunately I still get the error - Argument of type 'any' is not assignable to parameter of type 'never'.ts(2345) when using the new syntax.

outputDevice.push(new Gpio(23, {mode: Gpio.OUTPUT}));

fivdi, 2020-07-14:
Are you JavaScript / TypeScript beginner?
Please post a complete but short JavaScript program that can be used to reproduce the error.

IanW69, 2020-07-14:
I am fairly new to Typescript \ Javascript but converting one of my programs over from Python which uses the same "Array" principle with the gpiozero Python module.

A very basic program to illustrate the issue but I get the following console output when trying to compile: -

/home/pi/garage/node_modules/ts-node/src/index.ts:434
    return new TSError(diagnosticText, diagnosticCodes)
           ^
TSError: ⨯ Unable to compile TypeScript:
src/index.ts:6:19 - error TS2345: Argument of type 'any' is not assignable to parameter of type 'never'.

6 outputDevice.push(new Gpio(23, {mode: Gpio.OUTPUT}));
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/index.ts:7:19 - error TS2345: Argument of type 'any' is not assignable to parameter of type 'never'.

7 outputDevice.push(new Gpio(24, {mode: Gpio.OUTPUT}));
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/index.ts:8:19 - error TS2345: Argument of type 'any' is not assignable to parameter of type 'never'.

8 outputDevice.push(new Gpio(25, {mode: Gpio.OUTPUT}));
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/index.ts:13:18 - error TS2339: Property 'digitalWrite' does not exist on type 'never'.

13 outputDevice[id].digitalWrite(1);
                    ~~~~~~~~~~~~

    at createTSError (/home/pi/garage/node_modules/ts-node/src/index.ts:434:12)
    at reportTSError (/home/pi/garage/node_modules/ts-node/src/index.ts:438:19)
    at getOutput (/home/pi/garage/node_modules/ts-node/src/index.ts:578:36)
    at Object.compile (/home/pi/garage/node_modules/ts-node/src/index.ts:775:32)
    at Module.m._compile (/home/pi/garage/node_modules/ts-node/src/index.ts:858:43)
    at Module._extensions..js (internal/modules/cjs/loader.js:1158:10)
    at Object.require.extensions.<computed> [as .ts] (/home/pi/garage/node_modules/ts-node/src/index.ts:861:12)
    at Module.load (internal/modules/cjs/loader.js:986:32)
    at Function.Module._load (internal/modules/cjs/loader.js:879:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:71:12)
[nodemon] app crashed - waiting for file changes before starting...
 

index.js

// eslint-disable-next-line @typescript-eslint/no-var-requires
const Gpio = require('pigpio').Gpio;

const outputDevice = [];
outputDevice.push(new Gpio(23, {mode: Gpio.OUTPUT}));
outputDevice.push(new Gpio(24, {mode: Gpio.OUTPUT}));
outputDevice.push(new Gpio(25, {mode: Gpio.OUTPUT}));

const id = getRndInteger(0, 2);
console.log(`Turning on LED: ${id}`);
outputDevice[id].digitalWrite(1);

/*
Works when using variable method

const led = new Gpio(25, {mode: Gpio.OUTPUT});
led.digitalWrite(1)
*/

function getRndInteger(min, max) {
  return Math.floor(Math.random() * (max - min + 1) ) + min;
}

fivdi, 2020-07-14:
I'm not familiar with typescript and can't tell you if the sample code posted is valid TypeScript. It looks like valid JavaScript to me. This makes me wonder why you are compiling it with the TypeScript compiler and questioning if it's even necessary to do that.

Assuming that the code is in a file called index.js, what happens if you run the command `sudo node index.js`

IanW69, 2020-07-14:
I really appreciate the quick response / feedback 
It looks like an issue with Typescript as it compiles / runs okay in Javascript.  You are correct I don't need Typescript for this basic program but the larger module has some Typescript components to it.

fivdi, 2020-07-14:
Good to hear that it's working now with JavaScript. I guess you'll have to take a look at some tutorials to figure out what needs to be done for TypeScript. I'll go ahead and close the issue now as it isn't related to pigpio.

closed 2020-07-14 by fivdi