fivdi/pigpio #118: Does pigpio support compute module 3 ?

jinyistudio, 2020-08-31:
Does pigpio support compute module 3 ?

jinyistudio, 2020-09-03:
1. CM3 is ok. node .js with sudo
2. I need as root when i porting to NODE-RED as node. Do you know how to run as root in the NODE-RED ?

fivdi, 2020-09-03:
>1. CM3 is ok.

Thank you for confirming this to be the case.

>node .js with sudo

Yes, this is correct. `sudo` is required by the underlying pigpio C library.

>2. I need as root when i porting to NODE-RED as node. Do you know how to run as root in the NODE-RED ? 

Unfortunately I'm not familiar with Node-RED. I'd suggest asking this question at the Node-RED forum (https://discourse.nodered.org/)

fivdi, 2020-09-04:
There is no actionable for pigpio here so I'll go ahead and close the issue.

closed 2020-09-04 by fivdi