fivdi/pigpio #119: Proximity Sensor Bad Readings

evandavey, 2020-09-14:
Hi,
I am using the example HC-SR04 code on a Pi Zero W with trigger on pin 22 and echo on pin 27.  I am measuring a known distance of 38cm.  If I scp a large file, the distance suddenly drops to 22cm.
Any idea what might be going on?
Best Regards,
Evan

fivdi, 2020-09-25:
>  Any idea what might be going on?

I'm guessing here but perhaps the scp command is blocking the single CPU on the Pi Zero W for longer periods of time and preventing the pigpio C library from doing what it needs to do in a timely way.

I'd suggest opening an issue over at the [pigpio C library repository](https://github.com/joan2937/pigpio/issues) and asking if they can provide more information.

evandavey, 2020-09-25:
Wouldn't that mean it should be a bigger distance?  i.e. longer ping time rather than shorter?

fivdi, 2020-09-25:
Not necessarily. I'm not familiar with the internal implementation details of the pigpio C library. However, if I'm not mistaken it uses DMA (Direct Memory Access) to monitor inputs. I would assume that the scp command will also result in DMA being used. This could result in bus contention if both processes are competing for resources. This may in turn result in less distance being measured rather than more. I'm only guessing here, that's why it would be better to open an issue over at the pigpio C library repository as the people over there will know more about the subject that I do.

evandavey, 2020-09-25:
OK, thanks - not so much of an issue if only happening under load but we are monitoring an application where we are seeing a lot of false triggers that may or may not be related to this issue.  Do you know if a kernel module like https://github.com/torvalds/linux/blob/master/drivers/iio/proximity/srf04.c would be more reliable?

fivdi, 2020-09-25:
> Do you know if a kernel module like https://github.com/torvalds/linux/blob/master/drivers/iio/proximity/srf04.c would be more reliable?

I haven't used that module so I don't know.

evandavey, 2020-09-25:
I did have it working at one stage but it would read for a bit then ocassionally lock up (but didn't do much testing). It is otherwise more complicated than using your library so sub-optimal.  Thanks for your help, I might post over at pigpio but don't really want to get too deep down the rabbit hole if I can find a work-around.

closed 2020-09-25 by evandavey

fivdi, 2020-09-25:
Here's a link to an issue I opened a few years ago but had forgotten about that may provide some additional information https://github.com/joan2937/pigpio/issues/52. It's a fairly long issue. It's a fairly long issue. This comment is interesting https://github.com/joan2937/pigpio/issues/52#issuecomment-195288773.