fivdi/pigpio #124: Lacking permissions to  run pigpio.gpioInitialise();

JoshGElias, 2021-01-02:
I've just started learning the Raspberry Pi platform and am struggling to find a good library to interface with GPIO. This one is the most starred by far and seems to have a the features I'll need, mostly read/write on GPIO and PWM.

I've followed the installation instructions. Ensured that the pigpio daemon is installed on my Pi. However I still get this error whenever I try to initialize pigpio.

`2021-01-02 11:37:35 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

/home/pi/Projects/hydro/node_modules/pigpio/pigpio.js:54
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (/home/pi/Projects/hydro/node_modules/pigpio/pigpio.js:54:12)
    at new Gpio (/home/pi/Projects/hydro/node_modules/pigpio/pigpio.js:158:5)`

JoshGElias, 2021-01-02:
I made a symbolic link from my node and npm scripts to the /usr/bin/ so that I can run sudo but am still getting the error.

fivdi, 2021-01-02:
Duplicate of #114

> I made a symbolic link from my node and npm scripts to the /usr/bin/ directory so that I can run sudo but am still getting the error.

The sumbolic links are not needed. Just do the following on the command line:
```console
sudo node your-program-name.js
```
If this doesn't work please post the complete output of the program.

JoshGElias, 2021-01-02:
There's no node script in my usr/bin/ directory so without the symbolic link I can't run `sudo node <script>`. `If I try I get sudo; node; command not found`.

The complete output of the program is the following:
```
pi@raspberrypi:~/Projects/hydro $ sudo node ./build/rgb_led.js 
2021-01-02 11:56:30 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

/home/pi/Projects/hydro/node_modules/pigpio/pigpio.js:54
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (/home/pi/Projects/hydro/node_modules/pigpio/pigpio.js:54:12)
    at new Gpio (/home/pi/Projects/hydro/node_modules/pigpio/pigpio.js:158:5)
    at Object.<anonymous> (/home/pi/Projects/hydro/build/rgb_led.js:14:11)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12)
    at internal/main/run_main_module.js:17:47
```

fivdi, 2021-01-02:
Node.js appears to be installed on the system in such a way that user `root` doesn't have access to the Node.js executable. In such cases what I would do is run the following command:
```console
sudo $(which node) your-program-name.js
```
Can you remove the symbolic links and give this a try to see if it works?

JoshGElias, 2021-01-02:
I'm using nvm to manage my node versions. Although I still had this issue before I used nvm as well.
I have removed the symbolic link and tried running my script with `sudo /home/pi/.nvm/versions/node/v14.15.3/bin/node build/rgb_led.js `. Still getting the same error though.

```
pi@raspberrypi:~/Projects/hydro $ sudo /home/pi/.nvm/versions/node/v14.15.3/bin/node build/rgb_led.js 
2021-01-02 14:51:52 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

/home/pi/Projects/hydro/node_modules/pigpio/pigpio.js:54
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (/home/pi/Projects/hydro/node_modules/pigpio/pigpio.js:54:12)
    at new Gpio (/home/pi/Projects/hydro/node_modules/pigpio/pigpio.js:158:5)
    at Object.<anonymous> (/home/pi/Projects/hydro/build/rgb_led.js:14:11)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12)
    at internal/main/run_main_module.js:17:47
```

I really appreciate you helping me debug this :)

fivdi, 2021-01-02:
The error is coming from the [pigpio C library](https://github.com/joan2937/pigpio) rather than the pigpio Node.js module. I'm not sure why it is happening and I can't reproduce the issue locally on a Raspberry Pi.

There error is detected by the pigpio C library [here](https://github.com/joan2937/pigpio/blob/V71/pigpio.c#L7282-L7291), so there's something different about the file `/dev/mem` file on your system. What do you see if you run the command `ls -l /dev/mem `? This is what I see:

```console
pi@raspberrypi:~ $ ls -l /dev/mem 
crw-r----- 1 root kmem 1, 1 Jan  2 21:09 /dev/mem
pi@raspberrypi:~ $ 
```

Additional questions:
- Is there anything special about your system setup? For example, is Docker being used? Or is it a stock Raspberry Pi OS installation?
- Are you sure pigpiod is **NOT** running on the syetem?

JoshGElias, 2021-01-02:
I just needed to `chmod -R 755` the bin directory where the node script was installed to allow all users to run it. It's working now!
Thank you for the suport @fivdi!

fivdi, 2021-01-02:
Good to hear that it's working now 😄 
I'll go and close the issue.

closed 2021-01-02 by fivdi

Ikfreecharles, 2023-12-19:
> I just needed to `chmod -R 755` the bin directory where the node script was installed to allow all users to run it. It's working now! Thank you for the suport @fivdi!

I'm having the same issue right now. Can you kindly explain what you mean by this? How did you solve the issue?

m66g, 2024-02-18:
First of all, @fivdi thank you for your great application!
Second, it is highly recommended, not to install or run nodejs with administrator privileges (sudo), but uns nvm instead,

I get the same error as @JoshGElias.

@JoshGElias Please explain what you mean by

> I just needed to chmod -R 755 the bin directory where the node script was installed to allow all users to run it. It's working now!

The script is just a js file that is executed. What bin directory where the node script was installed do you mean?

@fivdi Any further hint on the error is of course highly appreciated as well.

Both of you, thank you very much in advance!

fivdi, 2024-02-18:
> First of all, @fivdi thank you for your great application!

@m66g Thank you and you are welcome 😄. That being said, the functionality is really coming from the [pigpio C library](https://github.com/joan2937/pigpio). The pigpio Node.js module is just a wrapper for the pigpio C library.

> Second, it is highly recommended, not to install or run nodejs with administrator privileges (sudo), but uns nvm instead,

I never install Node.js with administrator privileges (sudo) and never recommended that anyone do this 😄. I almost never run Node.js applications with administrator privileges and don't like doing so when I do it. However, the pigpio C library requires administrator privileges so I don't have a choice when using it.

> I get the same error as @JoshGElias.
>
> @JoshGElias Please explain what you mean by
>
>
> The script is just a js file that is executed. What bin directory where the node script was installed do you mean?

Although I'm not 100% sure, I would imagine that @JoshGElias means the directory containing the JavaScript file that was run and contains the code using pigpio.

> 
> @fivdi Any further hint on the error is of course highly appreciated as well.

If you don't want to use administrator privileges  you shouldn't use the Node.js pigpio module.