fivdi/pigpio #126: child_process.exec doesn't call callback when using pigpio in my project

J-Cat, 2021-01-17:
I switched to pigpio (vs onoff) to fix the issues I was having with skipped steps in the rotary encoder (which worked wonderfully by the way!), however, now, all my calls to child_process.exec never enter the callback.

If I swap back to using onoff, it's fine, but as soon as I use pigpio (using interrupts), it stops working.

fivdi, 2021-01-17:
Can you post a short but complete JavaScript program that can be used to reproduce the problem please?

J-Cat, 2021-01-17:
> Can you post a short but complete JavaScript program that can be used to reproduce the problem please?

Just did this, but no point in posting as it worked!  So there is something else ... it is very strange that everything else being the same (except swapping onoff for pigpio), and it works properly with one and not the other.  This is a quite large project though, so there is obviously something else causing the issue.

I'd spend more time on it, but the "work-around" (although I don't love it), solves the issue 100% and there are no other issues in the application that I know of (and I've been running it 24x7 for months

Thanks for the response!

closed 2021-01-17 by J-Cat

fivdi, 2021-01-18:
I wonder if the issue is related to signals. The pigpio C library makes fairly extensive usage of signals. The documentation for the pigpio Node.js module mentions this topic in the description of the [`initialization()` method](https://github.com/fivdi/pigpio/blob/master/doc/configuration.md#initialize).

J-Cat, 2021-01-20:
Yes, this was 100% the issue.  I added a call to initialize() and terminate() at the start of the app and on cleanup respectively, and now everything works exactly as expected.

Thank you very much for this!  (this was bothering me not knowing what the root cause of the issue was and instead just having a work-around)