fivdi/pigpio #127: sigHandler: Unhandled signal 2, terminating

kiko0217, 2021-02-02:
what is this problem? what is cause?

thank you for your answere

fivdi, 2021-02-05:
The message
```
sigHandler: Unhandled signal 2, terminating
```
is printed by the pigpio C library signal handler [here](https://github.com/joan2937/pigpio/blob/master/pigpio.c#L5628)

Signal 2 or SIGINT is the "terminal interrupt" signal that is sent when the user types the terminal interrupt character which is normally ctrl+c.

I guess you hit ctrl+c to terminate the application. If this is the case all is ok and there isn't a problem. To the best of my knowledge, the pigpio C library can't be configured to prevent it from handling SIGINT the way it does.

fivdi, 2021-02-07:
To the best of my knowledge nothing needs to be changed in pigpio here, so I'll go ahead and close the issue.

closed 2021-02-07 by fivdi

lwestfall, 2021-12-05:
For those of us who ended up here wanting to implement their own signal handler, make sure you do your `signal()` call AFTER you call `gpioInitialise()`. This will override the pigpio handler.

fivdi, 2021-12-05:
> For those of us who ended up here wanting to implement their own signal handler, make sure you do your `signal()` call AFTER you call `gpioInitialise()`. This will override the pigpio handler.

For further information and an example see the [documentation for the `initialize()` method](https://github.com/fivdi/pigpio/blob/master/doc/configuration.md#initialize).

fivdi, 2022-04-07:
> I am working on interfacing dht11 with Raspberry Pi. My code is as per this link: https://github.com/jimstudt/dht11-pigpio/blob/master/dht11-pigpio.c. My concern is when I press Ctrl+C, I get the error "Unhandled Signal 2, terminating" and after that I cannot write further commands on my linux terminal. Can anyone tell me what should I do?
> 
> [image]

@darshitagrawal This repository is the Node.js (JavaScript) pigpio repository. Support for the pigpio C library is available at https://github.com/joan2937/pigpio