fivdi/pigpio #128: Its possible to determine period of pulses using the tick value from on('alert') - like frecuency calculator

lucianobustos, 2021-02-06:
This is maybe a silly question but I want to measure the interval beetwen consecutive ticks. (aka. a frecuency calculator).
I've took this portion of your example... 

let endTick = 0;
let startTick = 0xffffffff;
btn.on('alert', (level, tick) => {
             endTick = tick;
             console.log((endTick >> 0) - (startTick >> 0)); // is this value acurrante to calculate the period??
             startTick = endTick;
});

I can apply some sampling techniques to get more accurate values and not on every "tick". But I want to know if this tick value count the interval beetwen 2 gpio.pulses/trigger...

Thanks.. this library it's amazing.

fivdi, 2021-02-06:
The tick value is an unsigned 32-bit integer value representing the number of microseconds since system boot. This value wraps around the 32-bit space in just over an hour.

As such, it should be possible to achieve what you would like to achieve.

See also [getTick()](https://github.com/fivdi/pigpio/blob/master/doc/global.md#gettick).

lucianobustos, 2021-02-06:
Thanks Brain again.
Actually I have a LM393 sensor with a sequencer-wheel to read pulses & convert them to freq. but doing that with node process.hrtime() is horrible and sensitive. (any proc. task corrupt the measurement) I'm testing your library and looks promising...

Thanks for your support.

closed 2021-02-06 by lucianobustos