fivdi/pigpio #129: Permission to pi user for hardware peripherals - how to make it work?

lucianobustos, 2021-02-11:
Hello I'm using this library but I can't run it without sudo command (as pi user). 
Which is the correct way to grant PI user permission to run pigpio inside node without sudo??

I've checked permisios, libs and I think the user has permission because, for example your fivdi/onoff lib works.
Is there any particular condition to meet for the pi user in order to run pigpio? 

> I was checking these 2 pages algo.
> https://roboticsbackend.com/raspberry-pi-hardware-permissions/#GPIOs
> https://www.raspberrypi.org/forums/viewtopic.php?t=8999

Thanks in advance!

lucianobustos, 2021-02-11:
I'm using pm2 to run node.

`sudo chmod -R 775 bin`  (where my start script is) solve the issue.

Maybe this could be somewhere in the docu.

fivdi, 2021-02-12:
I'm afraid it's not possible to run an app that uses the pigpio Node.js module without root permissions. The pigpio Node.js module is a wrapper for the [pigpio C library](https://github.com/joan2937/pigpio) and the C library requires root permissions.

fivdi, 2021-02-18:
I'll go ahead and close this as there is no TODO for pigpio.

closed 2021-02-18 by fivdi