fivdi/pigpio #131: Can't gpioInitialise on a Pi 400

fisherds, 2021-03-18:
I'm aware this library is still in beta for all Pi 4 Model B devices.  I just wanted to share that it works fine on my normal Pi 4 B (https://www.adafruit.com/product/4292), but not on my Pi 400 (https://www.amazon.com/dp/B08RMJR4CK/ref=twister_B08RMNBYBK).  Here is the error message I get on a Pi 400.  

```
pi@fisherds-pi400:~/github/spring-2021-personal-repo-fisherds/Node/Prep/GPIO/using_pigpio $ sudo node blink.js 
Ready
Setup pin 14 as an output
2021-03-18 08:58:36 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, this system does not appear to be a raspberry pi. |
|aborting.                                                |
+---------------------------------------------------------+

/home/pi/github/spring-2021-personal-repo-fisherds/Node/node_modules/pigpio/pigpio.js:54
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (/home/pi/github/spring-2021-personal-repo-fisherds/Node/node_modules/pigpio/pigpio.js:54:12)
    at new Gpio (/home/pi/github/spring-2021-personal-repo-fisherds/Node/node_modules/pigpio/pigpio.js:158:5)
    at runBlink (/home/pi/github/spring-2021-personal-repo-fisherds/Node/Prep/GPIO/using_pigpio/blink.js:26:17)
    at main (/home/pi/github/spring-2021-personal-repo-fisherds/Node/Prep/GPIO/using_pigpio/blink.js:21:5)
    at Object.<anonymous> (/home/pi/github/spring-2021-personal-repo-fisherds/Node/Prep/GPIO/using_pigpio/blink.js:38:1)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12)

```

If there is so way to use this library on a Pi 400, I'd love to get it figured out.  I'll have students test it in class today, so we'll have 20 more Pi 400 test users soon. :)  Let me know if there is anything you want me to try.  Thanks!

Note: for now we'll just use your onoff library on the Pi 400, but we eventually need features that are only available in your pigpio library.

fivdi, 2021-03-18:
What version of the pigpio C library is installed? Please check with the following command:
```bash
pigpiod -v
```
If the version is less than 74, please update to the latest version of the pigpio C library.
Installation instructions can be found [here](http://abyz.me.uk/rpi/pigpio/download.html).

See also https://github.com/joan2937/pigpio/issues/412

fisherds, 2021-03-19:
Yep, you nailed it.  The default version of pigpiod is 71 right now.  That does not work on the Pi 400.  I did the update and got 79, then it worked fine.  Thanks!

For my notes to remember what I did...
[ref](http://abyz.me.uk/rpi/pigpio/download.html)
```
pigpiod -v
(was 71)

sudo killall pigpiod
wget https://github.com/joan2937/pigpio/archive/master.zip
unzip master.zip
cd pigpio-master/
make
sudo make install
sudo pigpiod
(actually what I did was 'sudo reboot', but sudo pigpiod should've been plenty)

pigpiod -v
(now 79)
```

closed 2021-03-19 by fisherds