fivdi/pigpio #133: Electron & pigpio: gpioInitialise is not a function

SamBThor, 2021-05-22:
Config: Raspberry pi 4B, Pi OS, tested with node 11 to 14.17.0 (currently 14.17.0.) getting same version with node -v and sudo node -v, pigpio uninstalled and installed with apt-get, then nvm install, in the right order, also did npm rebuild and finally tried to install using with wget from https://github.com/joan2937/pigpio/archive/master.zip

HELP! 
I developed several little programs with pigpio a few months ago using node & electron. They worked fine.
Last week, I tried to execute them again and systematically got (and am still getting) the error below. I have read and followed the instructions of dozens of posts on this topic, to no avail. 
Unfortunately, I have done so many installs and upgrades on this Pi in the last months to set it up for my needs that it is not possible to determine what broke pigpio and I would really not like to start again from a clean install of Pi OS.
I am at a loss so I'll be very grateful for any help.
S.

> tests@1.0.0 test /home/pi/tests01
> electron .

+-----------------------------------------------------------------------+
| Warning: The pigpio C library can't be loaded on this machine and any |
| attempt to use it will fail.                                          |
|                                                                       |
| Error: "Module did not self-register"                                 |
| ------------------------------------                                  |
| If you are working on a Raspberry Pi and see a "Module did not self-  |
| register" error, this typically indicates that the installation       |
| instructions were not exactly followed. For further details see the   |
| installation section of the readme at                                 |
| https://github.com/fivdi/pigpio#installation                          |
| Note that step 1 of the installation instructions must be completed   |
| before step 2.                                                        |
+-----------------------------------------------------------------------+

Invoking require('bindings')('pigpio.node') resulted in the follwoing error:

Error: The module '/home/pi/tests01/node_modules/pigpio/build/Release/pigpio.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 83. This version of Node.js requires
NODE_MODULE_VERSION 85. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (electron/js2c/asar_bundle.js:5:1812)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1203:18)
    at Object.func [as .node] (electron/js2c/asar_bundle.js:5:1812)
    at Module.load (internal/modules/cjs/loader.js:992:32)
    at Module._load (internal/modules/cjs/loader.js:885:14)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12633)
    at Module.require (internal/modules/cjs/loader.js:1032:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at bindings (/home/pi/tests01/node_modules/bindings/bindings.js:112:48)
    at /home/pi/tests01/node_modules/pigpio/pigpio.js:8:31

-------------------------------------------------------------------------------

LED swipes.
App threw an error during load
TypeError: pigpio.gpioInitialise is not a function
    at initializePigpio (/home/pi/tests01/node_modules/pigpio/pigpio.js:54:12)
    at new Gpio (/home/pi/tests01/node_modules/pigpio/pigpio.js:158:5)
    at Object.<anonymous> (/home/pi/tests01/index.js:4:14)
    at Module._compile (internal/modules/cjs/loader.js:1152:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1173:10)
    at Module.load (internal/modules/cjs/loader.js:992:32)
    at Module._load (internal/modules/cjs/loader.js:885:14)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12633)
    at loadApplicationPackage (/home/pi/tests01/node_modules/electron/dist/resources/default_app.asar/main.js:110:16)
    at Object.<anonymous> (/home/pi/tests01/node_modules/electron/dist/resources/default_app.asar/main.js:222:9)
A JavaScript error occurred in the main process
Uncaught Exception:
TypeError: pigpio.gpioInitialise is not a function
    at initializePigpio (/home/pi/tests01/node_modules/pigpio/pigpio.js:54:12)
    at new Gpio (/home/pi/tests01/node_modules/pigpio/pigpio.js:158:5)
    at Object.<anonymous> (/home/pi/tests01/index.js:4:14)
    at Module._compile (internal/modules/cjs/loader.js:1152:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1173:10)
    at Module.load (internal/modules/cjs/loader.js:992:32)
    at Module._load (internal/modules/cjs/loader.js:885:14)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12633)
    at loadApplicationPackage (/home/pi/tests01/node_modules/electron/dist/resources/default_app.asar/main.js:110:16)
    at Object.<anonymous> (/home/pi/tests01/node_modules/electron/dist/resources/default_app.asar/main.js:222:9)
/home/pi/tests01/node_modules/electron/dist/electron exited with signal SIGSEGV
npm ERR! Test failed.  See above for more details.

fivdi, 2021-05-23:
I'm afraid I'm not familiar with Electron and don't know if I can provide much help. However, pigpio is a native Node module and special steps need to be taken when installing native Node modules with Electron. For additional information see [here](https://www.electronjs.org/docs/tutorial/using-native-node-modules).

fivdi, 2021-06-19:
There is no TODO for pigpio here so I'll go ahead and close the issue.

closed 2021-06-19 by fivdi