fivdi/pigpio #134: Error: The module '.../pigpio.node' was compiled against a different Node.js version

Zeprakadebra, 2021-06-19:
Hi fivdi,

when including your module into node.js I get the following error message:

0|mm     | Error: The module '/home/pi/MagicMirror/modules/MMM-PiTemp/node_modules/pigpio/build/Release/pigpio.node'
0|mm     | was compiled against a different Node.js version using
0|mm     | NODE_MODULE_VERSION 64. This version of Node.js requires
0|mm     | NODE_MODULE_VERSION 85. Please try re-compiling or re-installing

Can you please let me know how I can solve this issue?
Or can you please offer me a re-compiled version of your module against Node.js Version 85?

Kind regards,

Zebrakadebra

fivdi, 2021-06-19:
This issue is not related to pigpio. It's related to the version of Node.js that was used when pigpio was installed and the version of Node.js that is being used at runtime by applications using pigpio.

NODE_MODULE_VERSION 64 implies Node.js v10 was used when pigpio was installed.
There is no version of Node.js that uses NODE_MODULE_VERSION 85. See [here](https://nodejs.org/en/download/releases/).

Are you by any chance using Electron?
If so, see [Using Native Node Modules](https://www.electronjs.org/docs/tutorial/using-native-node-modules).

Zeprakadebra, 2021-06-21:
Hi fivdi,
thanks for your quick and quality reply. All good to know.

You are right: my code is embedded into electron, but to be honest i am still learning how to understand it.

So I am going to check your link regarding the native modules and then i let know my progress on that.

Kind regards,
Zebrakadebra

Zeprakadebra, 2021-06-23:
Hi fivdi,
The instructions for [Using Native Node Modules](https://www.electronjs.org/docs/tutorial/using-native-node-modules) worked perfectly for me. Thanks a lot for your hints. This issue can be closed.
Kind regards,
Zebrakadebra

fivdi, 2021-06-23:
@Zeprakadebra that's good News. Thanks for the feedback.

closed 2021-06-23 by fivdi