fivdi/pigpio #135: Unable to use package

Zebiano, 2021-07-20:
Hey! I'm currently facing the following issue:
```js
const Gpio = require('pigpio').Gpio;

const button = new Gpio(23, {
  mode: Gpio.INPUT,
  pullUpDown: Gpio.PUD_UP,
  alert: true
});
```
yields the following error:
```js
 mode: Gpio.INPUT,
             ^

TypeError: Cannot read property 'INPUT' of undefined
```
`console.log(Gpio)` returns undefined. Any idea as to why this is happening?

I'm using a Raspberry Pi 4 with Ubuntu Server 21.04 and Node 14.17.3. I've installed the `pigpio` C library, but I had to `make` it, as `sudo apt-get install pigpio` gave me an `E: Package 'pigpio' has no installation candidate` error. Running `pigpiod -v` results in `79`. I'm also running Node with `sudo`, since that is apparently a requirement, and I did run `npm i`, as well as `sudo npm i`.

Any help is greatly appreciated!

Zebiano, 2021-07-20:
Turns out I was an idiot and required the package wrongly... Instead of 
```js
const Gpio = require('pigpio').gpio;
```
It should be be:
```js
const Gpio = require('pigpio').Gpio;
```
This happened because I usually don't use `PascalCase`, but instead `camelCase`, so I blindly replaced *every* `Gpio` entry to `gpio`. 

Sorry for wasting an issue on this. It works now, I'm able to read data through the GPIO pins on the Raspberry Pi, thank you!

closed 2021-07-20 by Zebiano

fivdi, 2021-07-20:
Step 1 and Step 2 documented in the [Installation](https://github.com/fivdi/pigpio#installation) section of the readme must be performed in the correct order. That is, Step 1 must be performed before step 2. If you didn't run them in the correct order, remove the pigpio Node.js module and install it again. It should not be necessary to call "sudo npm install pigpio" to install the piogio Node.js module. Calling "npm install pigpio" should be enough.

>  I'm also running Node with sudo, since that is apparently a requirement

This is a requirement when running applications that use the pigpio Node.js module as the pigpio C library requires root privileges.