fivdi/pigpio #136: Consider relaxing restrictions in bindings.gyp

cinderblock, 2021-08-31:
Currently, `pigpiod` must be installed for this package to be built at all: https://github.com/fivdi/pigpio/blob/master/binding.gyp#L7

However, this is not a true requirement. If you want to be minimal, `pigpiod` need not even be installed.

Specifically, all that is needed is `libpigpio-dev` (`sudo apt-get install -y libpigpio-dev`).

It provides the necessary header and shared object files:

```log
pi@raspberrypi:~ $ dpkg -L libpigpio-dev
/.
/usr
/usr/include
/usr/include/pigpio.h                                          <----------
/usr/lib
/usr/share
/usr/share/doc
/usr/share/doc/libpigpio-dev
/usr/share/doc/libpigpio-dev/changelog.Debian.gz
/usr/share/doc/libpigpio-dev/copyright
/usr/share/man
/usr/share/man/man3
/usr/share/man/man3/pigpio.3.gz
/usr/lib/libpigpio.so                                          <----------
```

I can create a dummy `pigpiod` to make the build script work:
```bash
touch /usr/local/bin/pigpiod
chmod +x /usr/local/bin/pigpiod
```

Is there a reason this `condition` exists in `bindings.gyp`? It seems to me like this just makes the install silently fail and we're forced to see the big warning on startup about "Module did not self-register".

I realize I could have just followed the official instructions (**and** `sudo systemctl disable pigpiod`) to make this work. However, imho, if `pigpiod` isn't even installed in the first place, this should still work since it's relying on the C API.

Would you consider changing how these instructions are given?

IMHO, just removing the condition entirely would prefered. If the build fails, I don't want it to silently continue. A "`../src/pigpio.cc:2:10: fatal error: pigpio.h: No such file or directory`" error should get people going in the right direction.

If you really want to keep the condition, we could just check for the existence of `/usr/include/pigpio.h`. This would remove the `pigiod` dependency.

At the same time, I would also consider only listing `sudo apt-get install -y libpigpio-dev` as a prerequisite for setup.

fivdi, 2021-09-05:
There are quite a few developers who want to develop their software that depends on the `pigpio` Node.js module on a Mac or a non-Raspberry Pi Linux system. At some stage they will the deploy their software to a Raspberry Pi for testing. These people have the goal of doing as much of their development as possible on a non Raspberry Pi system where it is not possible to install the pigpio C library.

The [following](https://github.com/fivdi/pigpio/blob/master/binding.gyp#L7) condition in `binding.gyp` makes this possible:

```
'"<!((which pigpiod 2>/dev/null) || echo not_found)" != "not_found"'
```

If `pigpiod` is found, `pigpio.cc` which contains the native C++ code of the pigpio Node.js module will be compiled and added to the binary for the native pigpio Node.js module.

If `pigpiod` is not found, `pigpio.cc` will not be compiled and added to the binary for the native pigpio Node.js module.

To facilitate this feature some type of check is needed. In this case, the existance of `pigpiod` was chosen for the check.

> I can create a dummy `pigpiod` to make the build script work

This will only function on a Raspberry Pi that has the pigpio C library installed. It will not function on a Mac or non-Raspberry Pi Linux system where the pigpio C library is not installed.

> If you really want to keep the condition, we could just check for the existence of `/usr/include/pigpio.h`. This would remove the `pigiod` dependency.

A condition is needed to support development on any type of system. There are many ways to install the pigpio C library / pigpiod. This means that there is no guarantee that `pigpio.h` will be located at `/usr/include/pigpio.h`.

>At the same time, I would also consider only listing `sudo apt-get install -y libpigpio-dev` as a prerequisite for setup.

I would prefer to allow people to the install pigpio C library / pigpiod however they choose. I would prefer not to recommend `sudo apt-get install -y libpigpio-dev` as this will often not install the latest version. The latest version can be installed using the instructions at http://abyz.me.uk/rpi/pigpio/download.html.

cinderblock, 2021-09-06:
> There are quite a few developers who want to develop their software that depends on the `pigpio` Node.js module on a Mac or a non-Raspberry Pi Linux system. At some stage they will the deploy their [...]

You can always just get a copy of `pigpio.h` to develop against. That's all modern code checks actually need.

> The [following](https://github.com/fivdi/pigpio/blob/master/binding.gyp#L7) condition in `binding.gyp` makes this possible:
> 
> ```
> '"<!((which pigpiod 2>/dev/null) || echo not_found)" != "not_found"'
> ```

Yes, this is what I'm suggesting removing, and the `OS=="linux"` condition.

> If `pigpiod` is not found, `pigpio.cc` will not be compiled and added to the binary for the native pigpio Node.js module.

This is a problem, imho.

On systems that *could*, but *don't*, have `libpigpio-dev` installed, the build will seem to work, but there will be a runtime failure. Imho, this error should happen at **compile time**.

> To facilitate this feature some type of check is needed. In this case, the existence of `pigpiod` was chosen for the check.

This is my point. `pidpiod` is not the right check. `pigpiod` could, technically, be installed without the development headers.

> > I can create a dummy `pigpiod` to make the build script work
>
> This will only function on a Raspberry Pi that has the pigpio C library installed. It will not function on a Mac or non-Raspberry Pi Linux system where the pigpio C library is not installed.

So what? As long as you have `pigpio.h` you can develop against it with modern code editors. You just need the `.so` for the linker.

> A condition is needed to support development on any type of system. There are many ways to install the pigpio C library / pigpiod. This means that there is no guarantee that `pigpio.h` will be [...]

You're correct that `/usr/include/pigpio.h` is not a guaranteed location but it is possible to predict where `gcc` will look for `pigpio.h` and check those places. Still, this is a moot point if you just got rid of the check and let the compilation fail when `pigpio.h` is missing on the system.

Your point about a condition to help development is well taken. But, imho, if it impacts the runtime, it should not be used.
 
> I would prefer to allow people to the install pigpio C library / pigpiod however they choose. I would prefer not to recommend `sudo apt-get install -y libpigpio-dev` as this will often not install [...]

The official instructions say to `sudo apt-get install pigpio`, which depends on `libpigpio-dev`. Therefore, `libpigpio-dev` is the official latest stable version/package.

----

To show what I mean, I've made this basically work on Windows:

1. Clone this repo
2. Edit `bindings.gyp`
    - Remove the conditions
    - Add `headers` to include path
3. Put a copy of needed headers in a `headers` folder.
    - `pigpio.h`
    - `pthread.h`
    - `_ptw32.h`
    - `sched.h`
5. Create a dummy/mock library compiled for Windows
4. Run `npm i`. No errors. It builds just fine.

Therefore, `pigpiod` is not required.
I understand that it's a simple test, but it's extra and not needed.
Personally, I'd rather `pigpiod` binaries not even be installed on my system.

----

I realize that the steps I'm enumerating above to allow it to compile on other systems are a little much for some developers.

As an alternative, what if instead of only compiling `pigpio.cc` on known compatible systems, you only **don't** compile it when a certain "developer" file exists. The conditions could test for a file like `DEV_NO_BUILD` that is `.gitignore`d in the main directory. The `README.md` could explain to developers on systems that cannot install `libpigpio-dev` should create this file to disable compilation.

fivdi, 2021-09-16:
I don't think the pain is worth the gain. It won't be possible to handle any additional use cases after the suggested change. In reality, you can achieve what you would like to achieve today by creating a dummy `pigpiod`.

In addition, as described [here](http://abyz.me.uk/rpi/pigpio/download.html), the recommended way to install the latest version of the pigpio C library is with the commands listed below. Note that these commands always install pigpiod.

```bash
wget https://github.com/joan2937/pigpio/archive/master.zip
unzip master.zip
cd pigpio-master
make
sudo make install
```

Using `sudo apt-get install ...` to install the pigpio C library will often install an old version of the C library.

cinderblock, 2021-09-16:
IMHO, it's more painful that the software installs seemingly without error (until runtime) when it already has checks for the needed libraries at compile time.

Even worse, IMHO, the runtime check for `pigpio.node` doesn't actually error. It just prints to `stderr` and tries to continue and let's other undefined references be what really bring a program down.

IMHO, as soon as you notice an error, do something about it. Paving over the problem that you know will cause other issues later is just making it more complicated to fix.

Also note that `which` is not really a thing on Windows. I realize that `pigpio` isn't either, but it's just an extra headache for some developers.

PS, the link you sent also suggests using `sudo apt-get install ...` on Raspberry Pi. It does not that it *might* not been the latest version but that is par for the course in `apt` and not really a big deal. It's far easier (and faster) to use `apt` than it is do build it yourself, especially on a Pi.

fivdi, 2021-09-17:
> IMHO, it's more painful that the software installs seemingly without error (until runtime) when it already has checks for the needed libraries at compile time.
>
> Even worse, IMHO, the runtime check [...]

Things were done this way because some developers wanted to implement unit tests that ran on non-Raspberry Pi machines. On Mac or Linux machines if I remember correctly.

> Also note that `which` is not really a thing on Windows. I realize that `pigpio` isn't either, but it's just an extra headache for some developers.

Yes, this is true. This would be relatively easy to fix.

> PS, the link you sent also suggests using `sudo apt-get install ...` on Raspberry Pi. It does note that it _might_ not been the latest version but that is par for the course in `apt` and not really a big deal. It's far easier (and faster) to use `apt` than it is do build it yourself, especially on a Pi.

In my experience, it has very often been the case that `sudo apt-get install ...` installed an old version. The packages that are installed via `sudo apt-get install ...` are not maintained by the developers of the pigpio C library and have a different release cycle.

cinderblock, 2021-09-18:
> Things were done this way because some developers wanted to implement unit tests that ran on non-Raspberry Pi machines. On Mac or Linux machines if I remember correctly.

I understand this. IMHO, making the user experience (imho) worse (by not failing as early as possible) so that the developer experience is ever so arguably better is not a good trade.

To wit, the only thing I can see that is "better" for developers is that they can run the command `node-gyp rebuild` and have it not fail. This doesn't seem necessary to me for developing on non-Raspberry Pi systems.

If a developer...
- needs real tests
  - They have a Pi, right?
- is working on or running tests on only the JavaScript
  - Why `node-gyp rebuild`?
  - Write tests that are ok with failures when `require`ing the `.node` file
- is working on C/C++
  - Most code editors code analysis tools only need the `.h` files
  - How will tests run if they aren't on a Pi? Stub library?

I feel like I'm missing something.
 
----

> In my experience, it has very often been the case that `sudo apt-get install ...` installed an old version. The packages that are installed via `sudo apt-get install ...` are not maintained by the developers of the pigpio C library and have a different release cycle.

This is true for many packages. However, in my experience, Debian's `pigpio` is rarely far behind `master`.

But I think this point is moot. The official instructions for this node package don't say to install from source, it says "`sudo apt-get install pigpio`". Changing that to `sudo apt-get install libpigpio-dev` (or just listing it as an alternative) seems simple enough, as long as the `pigpiod` check is removed.

fivdi, 2021-09-18:
> I feel like I'm missing something.

This may be the case. I have tried to explain why decisions were made the way they were but apparently not in a comprehensible enough way.

closed 2021-09-18 by fivdi

cinderblock, 2021-09-18:
Is there some particular consideration you're making for the developers that you think I'm not seeing? Did I miss anything in my list of reasons that this _is_ helpful for developers?