fivdi/pigpio #139: Error: EINVAL, Invalid argument

samwelkinuthia, 2021-11-18:
I am using pigpio for servo control and i get the following error

[image]

fivdi, 2021-11-18:
In your code, on line 32 of server.js, there is a call to `servoWrite`. As described [here in the documentation](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#servowritepulsewidth), the value passed to `servoWrite` must be an unsigned integer, that is, the value passed to `servoWrite` must be 0 or a number in the range 500 through 2500. Is your code calling `servoWrite` correctly? It looks like it isn't.

samwelkinuthia, 2021-11-19:
Thank you! I was passing it incorrectly.

closed 2021-11-19 by samwelkinuthia