fivdi/pigpio #140: how can i read data from gps from pin 15 tx 16 rx ?

alimuhnad, 2021-11-29:


fivdi, 2021-11-29:
The bitbanged UART functionality provided by the pigpio C library is not available in the pigpio Node.js module.

Duplicate of #40.

alimuhnad, 2021-11-29:
so what can i do ???? and idea? i have another problem with using servo i am donig drone in js and i am using this lib to controll on the motors so the problem is that when i send the pulse width to the motors on the pins 16 and 20 26 and 21 the motors respons slow not fast..... i dont understand where is the problem with lib or my code or pigpio config?
thanks for respons on me

fivdi, 2021-11-29:
> so what can i do ???? and idea?

A Raspberry Pi Zero, 1, 2 and 3 has two UARTs and a Raspberry Pi 4 has six UARTs. (See [here](https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-uarts).) I would use one of those UARTs to communicate with the GPS module and access the UART using [serialport](https://github.com/serialport/node-serialport) rather than using bitbanged solution.

> i have another problem with using servo 

Please don't discuss multiple unrelated topics in the same issue. If there is a problem with servos in the pigpio Node.js library please open a new issue describing what the problem is and provide a small but complete example program that can be used to reproduce the problem.

alimuhnad, 2021-11-29:
ok  ill see about the gps now its work using gps lib and SerialPort ......... you can add this option to the pigpio it will be good thing to developers ..... thanks :)

closed 2021-11-29 by alimuhnad