fivdi/pigpio #141: using pigpio with 4 motors

alimuhnad, 2021-12-02:
hey i am trying to using pigpio with 4 motors using pins 20 21 26 16 the problem is that its not fast enough i need to set up the frequencies  and the dutyCycle  for thes pins 
for those pins pwm set up functions  will not work i try it out 
and i try this function its give me error hardwarePwmWrite can you give me example with numbers
- the time of executeing the pulse width must be in micro sec 
- the main problem is that i am trying to balance x axis using pid controller

fivdi, 2021-12-02:
Hardware PWM is not available on pins 20, 21, 26 or 16. Attempting to call `hardwarePwmWrite` for these pins will result in errors.

On a Raspberry Pi 4 B, you only have two usable hardware PWM pins, for further details https://forums.raspberrypi.com/viewtopic.php?t=308908#p1848068. The function [hardwarePwmWrite](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#hardwarepwmwritefrequency-dutycycle) can only be used for these two pins.

DMA based PWM is available on any GPIO. For DMA based GPIO used the functions [pwmFrequency(frequency)](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#pwmfrequencyfrequency), [pwmRange(range)](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#pwmrangerange) and [pwmWrite(dutyCycle)](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#pwmwritedutycycle).

If something is not functioning correctly, please post a complete but short sample program that can be used to reproduce the problem.

alimuhnad, 2021-12-02:
so you saying that DMA based PWM is available on any GPIO that means i can use the functions pwmFrequency(frequency), pwmRange(range) and pwmWrite(dutyCycle). on  pins 20, 21, 26 or 16.? i tryed but there is no Effect on the pins

fivdi, 2021-12-02:
> so you saying that DMA based PWM is available on any GPIO that means i can use the functions pwmFrequency(frequency), pwmRange(range) and pwmWrite(dutyCycle). on pins 20, 21, 26 or 16.?

Yes.

> i tryed but there is no Effect on the pins

As already mentioned above, if something is not functioning correctly, please post a complete but short sample program that can be used to reproduce the problem.

Also, post a photo of your Raspberry Pi GPIO header where pins 20, 21, 26 and 16 are marked so that I can see them.

alimuhnad, 2021-12-07:
after a lot of seaches i find that i need to make 4 motor work on this (Waveforms to generate GPIO level changes (time accurate to a few µs)) now how can i  use waveforms .......... can you plz give me example 
this is my code in servo but its not fast 
const pigpio = require('pigpio');
pigpio.configureClock(10, pigpio.CLOCK_PWM)
const Gpio = pigpio.Gpio;
// set gpio pins 
const motor1 = new Gpio(20, { mode: Gpio.OUTPUT });
const motor2 = new Gpio(26, { mode: Gpio.OUTPUT });
const motor3 = new Gpio(21, { mode: Gpio.OUTPUT });
const motor4 = new Gpio(16, { mode: Gpio.OUTPUT });

function speeds(m1, m2, m3, m4) {
  if (m1 > 1450) {
    m1 = 1450;
  }
  if (m2 > 1450) {
    m2 = 1450
  }
  if (m3 > 1450) {
    m3 = 1450
  }
  if (m4 > 1450) {
    m4 = 1450
  }
  if (m1 < 1185) {
    m1 = 1185;
  }
  if (m2 < 1185) {
    m2 = 1185
  }
  if (m3 < 1185) {
    m3 = 1185
  }
  if (m4 < 1185) {
    m4 = 1185
  }
 
  motor1.servoWrite(m1)
   motor2.servoWrite(m2)
  motor3.servoWrite(m3)
  motor4.servoWrite(m4)
}

fivdi, 2021-12-07:
The code posted above doesn't do very much. It only creates four `Gpio` objects but it doesn't do anything with them. After creating the four `Gpio` objects the program terminates immediately. The code can not be used to reproduce a problem.

The readme has an example that shows how to generate a waveform.

If your application is running too slowly, then you will need to figure out why it is running too slowly. Switching from using the `servoWrite` method to using waveforms is not going to make your program faster.

alimuhnad, 2021-12-07:
Mmmm this is sooooo weird if weveform same speed of servowrite ....... Can you give your WhatsApp so i can explain to you i and show you whats happing my be here i make its not clear enough

fivdi, 2021-12-07:
That's not something I can do.