fivdi/pigpio #142: High CPU usage on Pi 4B

kobuki, 2022-05-01:
I'm running this simple test app:

```
const Gpio = require('pigpio').Gpio;
const led = new Gpio(17, {mode: Gpio.OUTPUT})
setTimeout(() => {
}, 15000)
```

It constantly uses around 7% CPU. If I comment out the `const led ...` line, the CPU usage of the app is unnoticeable, normal. Using RasPi OS Bullseye, kernel 5.15.32-v7l+, 32 bits. If I add more Gpio instances (for simple RGB LED controlling), the usage doesn't change noticeably, even if I actually control them, i. e. change PWM duty cycles, etc.

Any hints?

closed 2022-05-01 by kobuki

fivdi, 2022-05-02:
The CPU usage seen here is related to how the pigpio C library in implemented. Additional information can be found at https://github.com/joan2937/pigpio/issues/29 or https://github.com/joan2937/pigpio/issues/206

kobuki, 2022-05-02:
Thanks, I realized that so I already closed the issue. Your library is pretty easy to use and lots of functions, it would be a pain to use something else. So I wonder why can't we have lower sample rates. For something like detecting button/switch input and the likes, even an 1 kHz sample rate would be more than adequate. Some ESP8266 and similar devices use as low a sample rate for such things as 100 Hz.

fivdi, 2022-05-02:
> So I wonder why can't we have lower sample rates. For something like detecting button/switch input and the likes, even an 1 kHz sample rate would be more than adequate. Some ESP8266 and similar devices use as low a sample rate for such things as 100 Hz.

The sample rates available in the pigpio Node.js module are the same as the sample rates available in the pigpio C library. The sample rate can be configured with function [configureClock](https://github.com/fivdi/pigpio/blob/master/doc/configuration.md#configureclockmicroseconds-peripheral). The corresponding pigpio C library function is documented [here](http://abyz.me.uk/rpi/pigpio/cif.html#gpioCfgClock). Only six sample rates are available ranging from 100,000 to 1,000,000 samples per second.

kobuki, 2022-05-02:
Yes, I know, I wanted to say it would be nice to have lower sample rates for simple functions without the need to drop your lib. Using `configureClock(10)` which still requires about 5% of a single core which is, considering the task, seems like a lot.

fivdi, 2022-05-02:
As mentioned above, the sampling rates available are determined by the pigpio C library, not the pigpio Node.js library. Adding new sampling rates would require changes to the [pigpio C library](https://github.com/joan2937/pigpio). Feel free to create an issue there if you have a valid use case. Alternatively, feel free to drop this lib and use an alternative.

kobuki, 2022-05-02:
Ah, sorry, i don't know why I referred to this lib as the C original yours wraps. Maybe those links confused me while I responded here between 2 other tasks.