fivdi/pigpio #145: Could not locate bindings file

nexuspcs, 2022-07-07:
Hi when ever I use node to run my JS file I am displayed this error:

```
doorpi@doorpi:~ $ cd Smartphone-Doorlock/
doorpi@doorpi:~/Smartphone-Doorlock $ node doorlock.js 
/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/node_modules/bindings/bindings.js:91
  throw err
  ^

Error: Could not locate the bindings file. Tried:
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/Debug/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/Release/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/out/Debug/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/Debug/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/out/Release/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/Release/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/default/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/compiled/16.15.1/linux/arm/pigpio.node
    at bindings (/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/node_modules/bindings/bindings.js:88:9)
    at Object.<anonymous> (/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/pigpio.js:5:31)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/doorpi/Smartphone-Doorlock/doorlock.js:47:12)
    at Module._compile (node:internal/modules/cjs/loader:1105:14) {
  tries: [
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/Debug/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/Release/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/out/Debug/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/Debug/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/out/Release/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/Release/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/default/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/compiled/16.15.1/linux/arm/pigpio.node'
  ]
}
doorpi@doorpi:~/Smartphone-Doorlock $ 
```

**Attempting to run using the `sudo $(which node)` gives me the same error:** 

```
doorpi@doorpi:~/Smartphone-Doorlock $ sudo $(which node) doorlock.js 
/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/node_modules/bindings/bindings.js:91
  throw err
  ^

Error: Could not locate the bindings file. Tried:
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/Debug/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/Release/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/out/Debug/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/Debug/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/out/Release/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/Release/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/default/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/compiled/16.15.1/linux/arm/pigpio.node
    at bindings (/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/node_modules/bindings/bindings.js:88:9)
    at Object.<anonymous> (/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/pigpio.js:5:31)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/doorpi/Smartphone-Doorlock/doorlock.js:47:12)
    at Module._compile (node:internal/modules/cjs/loader:1105:14) {
  tries: [
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/Debug/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/Release/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/out/Debug/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/Debug/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/out/Release/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/Release/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/default/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/compiled/16.15.1/linux/arm/pigpio.node'
  ]
}
doorpi@doorpi:~/Smartphone-Doorlock $
```

My node versions are below:

`node -v`: v16.15.1
`sudo node -v`: v16.15.1

Any assistance would be deeply thanked and welcome, thank you, !

nexuspcs, 2022-07-08:
> It looks like you forgot step 2 of the installation instructions.

HI, I have tried running this command as well and I am greeted with this error/output:

``` 
doorpi@doorpi:~ $ npm install pigpio
npm ERR! code EACCES
npm ERR! syscall open
npm ERR! path /home/doorpi/package-lock.json
npm ERR! errno -13
npm ERR! Error: EACCES: permission denied, open '/home/doorpi/package-lock.json'
npm ERR!  [Error: EACCES: permission denied, open '/home/doorpi/package-lock.json'] {
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'open',
npm ERR!   path: '/home/doorpi/package-lock.json'
npm ERR! }
npm ERR! 
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR! 
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/doorpi/.npm/_logs/2022-07-08T00_15_24_298Z-debug-0.log
doorpi@doorpi:~ $ 
```

it sounds like a sudo error maybe, so I ran `sudo npm install pigpio`, and I was presented this output:
```
doorpi@doorpi:~ $ sudo npm install pigpio

changed 4 packages, and audited 5 packages in 27s

found 0 vulnerabilities
doorpi@doorpi:~ $ 
```

However, the JS file I want to run still produces the same errors. ::

```
doorpi@doorpi:~ $ cd Smartphone-Doorlock/
doorpi@doorpi:~/Smartphone-Doorlock $ node doorlock.js 
/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/node_modules/bindings/bindings.js:91
  throw err
  ^

Error: Could not locate the bindings file. Tried:
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/Debug/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/Release/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/out/Debug/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/Debug/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/out/Release/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/Release/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/default/pigpio.node
 → /home/doorpi/Smartphone-Doorlock/node_modules/pigpio/compiled/16.15.1/linux/arm/pigpio.node
    at bindings (/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/node_modules/bindings/bindings.js:88:9)
    at Object.<anonymous> (/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/pigpio.js:5:31)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/doorpi/Smartphone-Doorlock/doorlock.js:47:12)
    at Module._compile (node:internal/modules/cjs/loader:1105:14) {
  tries: [
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/Debug/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/Release/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/out/Debug/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/Debug/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/out/Release/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/Release/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/build/default/pigpio.node',
    '/home/doorpi/Smartphone-Doorlock/node_modules/pigpio/compiled/16.15.1/linux/arm/pigpio.node'
  ]
}
doorpi@doorpi:~/Smartphone-Doorlock $ 
```

Thank you for your time, any other suggestions?

fivdi, 2022-07-08:
The problems that are occurring are not related to pigpio, they are related to Linux permissions. Running commands with "sudo" that shouldn't have been run with "sudo" has resulted in directories been created that are owned by root. User doorpi can not  access the directories owned by root and this results in EACCES errors. The attempts to install pigpio with sudo has probably messed things on your system up a little.

The following works for me, give it a try to see if it works for us:

```
pi@raspberrypi:~ $ cd # change to home directory
pi@raspberrypi:~ $ pigpiod -v
79
pi@raspberrypi:~ $ mkdir myapp
pi@raspberrypi:~ $ cd myapp
pi@raspberrypi:~/myapp $ npm install pigpio

added 4 packages, and audited 5 packages in 18s

found 0 vulnerabilities
npm notice 
npm notice New minor version of npm available! 8.12.1 -> 8.13.2
npm notice Changelog: https://github.com/npm/cli/releases/tag/v8.13.2
npm notice Run npm install -g npm@8.13.2 to update!
npm notice 
pi@raspberrypi:~/myapp $ sudo node node_modules/pigpio/example/gpio-info.js 
GPIO 0: mode=0 level=1
GPIO 1: mode=0 level=1
GPIO 2: mode=0 level=1
GPIO 3: mode=0 level=1
GPIO 4: mode=0 level=1
GPIO 5: mode=0 level=1
GPIO 6: mode=0 level=1
GPIO 7: mode=0 level=1
GPIO 8: mode=0 level=1
GPIO 9: mode=0 level=0
GPIO 10: mode=0 level=0
GPIO 11: mode=0 level=0
GPIO 12: mode=0 level=0
GPIO 13: mode=0 level=0
GPIO 14: mode=4 level=1
GPIO 15: mode=4 level=1
GPIO 16: mode=0 level=0
GPIO 17: mode=0 level=0
GPIO 18: mode=0 level=0
GPIO 19: mode=0 level=0
GPIO 20: mode=0 level=0
GPIO 21: mode=0 level=0
GPIO 22: mode=0 level=0
GPIO 23: mode=0 level=0
GPIO 24: mode=0 level=1
GPIO 25: mode=0 level=1
GPIO 26: mode=0 level=0
GPIO 27: mode=0 level=0
GPIO 28: mode=2 level=1
GPIO 29: mode=2 level=0
GPIO 30: mode=0 level=1
GPIO 31: mode=0 level=1
GPIO 32: mode=0 level=1
GPIO 33: mode=0 level=1
GPIO 34: mode=7 level=1
GPIO 35: mode=7 level=1
GPIO 36: mode=7 level=1
GPIO 37: mode=7 level=1
GPIO 38: mode=7 level=1
GPIO 39: mode=7 level=1
GPIO 40: mode=4 level=1
GPIO 41: mode=4 level=0
GPIO 42: mode=1 level=0
GPIO 43: mode=0 level=1
GPIO 44: mode=0 level=1
GPIO 45: mode=0 level=1
GPIO 46: mode=0 level=0
GPIO 47: mode=0 level=0
GPIO 48: mode=0 level=0
GPIO 49: mode=0 level=0
GPIO 50: mode=0 level=0
GPIO 51: mode=0 level=0
GPIO 52: mode=0 level=0
GPIO 53: mode=0 level=0
pi@raspberrypi:~/myapp $ 
```

nexuspcs, 2022-07-08:
Hi, when I run those commands above, I get this output/error:

``` oorpi@doorpi:~/Smartphone-Doorlock $ cd
doorpi@doorpi:~ $ pigpio -v
-bash: pigpio: command not found
doorpi@doorpi:~ $ pigpiod -v 
79
doorpi@doorpi:~ $ mkdir myapp
doorpi@doorpi:~ $ cd myapp/
doorpi@doorpi:~/myapp $ npm install pigpio
npm ERR! code EACCES
npm ERR! syscall open
npm ERR! path /home/doorpi/package-lock.json
npm ERR! errno -13
npm ERR! Error: EACCES: permission denied, open '/home/doorpi/package-lock.json'
npm ERR!  [Error: EACCES: permission denied, open '/home/doorpi/package-lock.json'] {
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'open',
npm ERR!   path: '/home/doorpi/package-lock.json'
npm ERR! }
npm ERR! 
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR! 
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/doorpi/.npm/_logs/2022-07-08T02_20_53_427Z-debug-0.log
doorpi@doorpi:~/myapp $ 
```

fivdi, 2022-07-08:
What is the complete output of the command `ls -l ~/myapp/`?

nexuspcs, 2022-07-08:
The complete output of the command `ls -l ~/myapp/` is: 
```
doorpi@doorpi:~ $ ls -l ~/myapp/
total 0
doorpi@doorpi:~ $ 
```

fivdi, 2022-07-08:
What is the complete output of the command `ls -l ~`?

nexuspcs, 2022-07-08:
`ls -l ~` outputs:

```
doorpi@doorpi:~ $ ls -l ~
total 3004
drwxr-xr-x 2 doorpi doorpi    4096 Apr  4 21:57 Bookshelf
drwxr-xr-x 2 doorpi doorpi    4096 Apr  4 22:46 Desktop
drwxr-xr-x 2 doorpi doorpi    4096 Apr  4 22:46 Documents
drwxr-xr-x 2 doorpi doorpi    4096 Apr  4 22:46 Downloads
-rw-r--r-- 1 doorpi doorpi 2993705 Jul  7 21:05 master.zip
drwxr-xr-x 2 doorpi doorpi    4096 Apr  4 22:46 Music
drwxr-xr-x 2 doorpi doorpi    4096 Jul  8 12:20 myapp
drwxr-xr-x 6 doorpi doorpi    4096 Jul  8 10:17 node_modules
-rw-r--r-- 1 doorpi doorpi   13764 Jul  7 20:56 nodesetup.sh
-rw-r--r-- 1 root   root        51 Jul  8 10:17 package.json
-rw-r--r-- 1 root   root      2610 Jul  8 10:17 package-lock.json
drwxr-xr-x 2 doorpi doorpi    4096 Apr  4 22:46 Pictures
drwxr-xr-x 7 doorpi doorpi    4096 Jul  7 21:07 pigpio-master
drwxr-xr-x 2 doorpi doorpi    4096 Apr  4 22:46 Public
drwxr-xr-x 7 doorpi doorpi    4096 Jul  7 22:20 quick2wire-gpio-admin
drwxr-xr-x 4 doorpi doorpi    4096 Jul  8 12:17 Smartphone-Doorlock
drwxr-xr-x 2 doorpi doorpi    4096 Apr  4 22:46 Templates
drwxr-xr-x 2 doorpi doorpi    4096 Apr  4 22:46 Videos
doorpi@doorpi:~ $ 
```

fivdi, 2022-07-08:
The files 'package.json' and 'package-lock.json' in your home directory are owned by root and are making a mess of things. Remove them.

Also, there is a directory called 'node_modules' in your home directory. Remove it completely (`sudo rm -rf node_modules`)

Then, try what I suggested at https://github.com/fivdi/pigpio/issues/145#issuecomment-1178465114 again.

nexuspcs, 2022-07-08:
Hi, that seems to have changed my errors:

```
doorpi@doorpi:~/myapp $ ls
doorlock.js  node_modules  package.json  package-lock.json
doorpi@doorpi:~/myapp $ node doorlock.js 
2022-07-08 12:48:16 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

/home/doorpi/myapp/node_modules/pigpio/pigpio.js:54
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (/home/doorpi/myapp/node_modules/pigpio/pigpio.js:54:12)
    at new Gpio (/home/doorpi/myapp/node_modules/pigpio/pigpio.js:158:5)
    at Object.<anonymous> (/home/doorpi/myapp/doorlock.js:48:11)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:77:12)
    at node:internal/main/run_main_module:17:47
doorpi@doorpi:~/myapp $ 
```
I think we are getting somewhere, but there are new errors which I am getting now

fivdi, 2022-07-08:
It should be `sudo node doorlock.js` and not `node doorlock.js `.

nexuspcs, 2022-07-08:
That still produces similar output: 

```
doorpi@doorpi:~/myapp $ sudo node doorlock.js
2022-07-08 12:52:06 initInitialise: Can't lock /var/run/pigpio.pid
/home/doorpi/myapp/node_modules/pigpio/pigpio.js:54
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (/home/doorpi/myapp/node_modules/pigpio/pigpio.js:54:12)
    at new Gpio (/home/doorpi/myapp/node_modules/pigpio/pigpio.js:158:5)
    at Object.<anonymous> (/home/doorpi/myapp/doorlock.js:48:11)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:77:12)
    at node:internal/main/run_main_module:17:47
doorpi@doorpi:~/myapp $ 
```

fivdi, 2022-07-08:
What is the output of command `ps -A | grep pigpiod`?

nexuspcs, 2022-07-08:
Produces: 
```
doorpi@doorpi:~/myapp $ ps -A | grep pigpiod
 1562 ?        00:00:02 pigpiod
doorpi@doorpi:~/myapp $ 
```

fivdi, 2022-07-08:
The pigpiod service is running. It shouldn't be running. Did you start it intentionally?
If you didn't start it intentionally run the command `sudo kill -9 1562` and try to run your node application again.

nexuspcs, 2022-07-08:
> The pigpiod service is running. It shouldn't be running. Did you start it intentionally?
> If you didn't start it intentionally run the command `sudo kill -9 1562` and try to run your node application again.

Hi, that has worked, now I am having some issues with SSL auth in blynk.:

```
doorpi@doorpi:~/myapp $ sudo node doorlock.js 

    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/

  Give Blynk a Github star! => https://github.com/vshymanskyy/blynk-library-js

Connecting to: blynk-cloud.com 443
locking door
SSL authorization...
SSL not authorized
Connecting to: blynk-cloud.com 443
SSL authorization...
SSL not authorized
```

fivdi, 2022-07-08:
I can't help you with SSL auth issues in blynk. The problems with the pigpio Node.js modules have been resolved. I'll go ahead and close this issue. Good luck.

closed 2022-07-08 by fivdi

nexuspcs, 2022-07-08:
Ok all good. 
Thank you so much for your help with pigpio !
Ill do some more research own ssl