fivdi/pigpio #146: pigpio.gpioInitialise() in pi 4B

alanezze, 2022-08-27:
hello, I'm trying to run by npm and node 14 from a raspberry and throws this error

``sevnt@raspberrypi:~/Documents/wiegand-rasberry $ sudo node send-wiegand.js 
2022-08-27 17:48:24 initInitialise: Can't lock /var/run/pigpio.pid
/home/sevnt/Documents/wiegand-rasberry/node_modules/pigpio/pigpio.js:54
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (/home/sevnt/Documents/wiegand-rasberry/node_modules/pigpio/pigpio.js:54:12)
"

Version NodeJs
v14.20.0

Version piGpio
79

check the previous problems resolved and none have worked

the only difference is that when checking the version of linux it is 00000D00
instead of C,
It is a newly purchased latest model raspberry with the latest and updated raspbian OS

nexuspcs, 2022-08-27:
run this command, it worked for me: `sudo killall pigpiod`

nexuspcs, 2022-08-27:
and then try to run this command, it may not work though: `sudo killall pigpio`

nexuspcs, 2022-08-27:
let me know how you go, I have a lot of issues as well

alanezze, 2022-08-27:
@nexuspcs 
Hello, I already tried restarting the PIGPIO daemon, and it's still the same

nexuspcs, 2022-08-27:
so you have tried: `sudo killall pigpiod`
maybe try this:; http://abyz.me.uk/rpi/pigpio/faq.html#Cant_initialise_pigpio_library

nexuspcs, 2022-08-27:
As this link describes run these commands:: 
- `cat /var/run/pigpio.pid` 
- `sudo kill -9 pid` (replace 'pid' with the id/pid number of the process printed out in the 1st command)

nexuspcs, 2022-08-27:
If this does not work, try this commands:

`sudo rm /var/run/pigpio.pid`

nexuspcs, 2022-08-27:
And then restart daemon: `sudo pigpiod`
And then try your script, and then if it still not works, try to kill: `sudo killall pigpiod` and `sudo killall pigpio`

nexuspcs, 2022-08-27:
I had this issue before, and it is what worked for me.   ....

nexuspcs, 2022-08-27:
#145 
This happened to me in this issue.

alanezze, 2022-08-27:
It worked but with this sequence:

`cat /var/run/pigpio.pid`
(save number)
`sudo rm /var/run/pigpio.pid`
`sudo killall pigpiod`

delete nodemodules and
`npm start`

muchas gracias por tu ayuda @nexuspcs

closed 2022-08-27 by alanezze

nexuspcs, 2022-08-27:
You are most welcome, I had similar issues, so I am glad that I could helps you !