fivdi/pigpio #150: "pigpio" error Maximum call stack size exceeded

Marco-exports, 2023-03-11:
Hi . . .

I followed all you install instructions with v79, but I keep getting :

verbose Linux 5.10.103-v7l+
24443 verbose argv "/home/pi/.nvm/versions/node/v12.22.12/bin/node" "/home/pi/.nvm/versions/node/v12.22.12/bin/npm" "install" "pigpio"
24444 verbose node v12.22.12
24445 verbose npm  v6.14.16
"pigpio" error Maximum call stack size exceeded

Marco-exports, 2023-03-11:
This error happens each time I do -->>   npm install pigpio

fivdi, 2023-03-12:
Following the installation instructions shown [here](https://github.com/fivdi/pigpio#installation), pigpio installs successfully for me:
```
$ npm install pigpio

added 4 packages, and audited 5 packages in 13s

found 0 vulnerabilities
$ 
```
Here is some additional information related to the Raspberry Pi being used:
```
$ uname -a
Linux raspberrypi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
$ node --version
v12.22.12
$ npm --version
7.5.2
$ pigpiod -v
79
$ 
```
I'm afraid I can't reproduce the error and assume the issue is related to how your system is set up. From the path `/home/pi/.nvm/versions/node/v12.22.12/bin/node` shown in the initial post above, the directory name `.nvm` indicates that you may be using nvm (Mode Version manager) which I'm not using. I can't explain why the error is occurring.

closed 2023-07-30 by fivdi