fivdi/pigpio #151: Unable to start pigpio on RaspberryPi 3B+ - bullseye

basil-dsouza, 2023-07-29:
Hello,

I have tried the below steps on a fresh install of Bullseye 64bit OS and am getting the below error (see at bottom for full stack trace)
| Warning: The pigpio C library can't be loaded on this machine and any |
| attempt to use it will fail.                                          |
|                                                                       |
| Error: "Module did not self-register"                                 |

Invoking require('bindings')('pigpio.node') resulted in the follwoing error: Error: Could not locate the bindings file. Tried:

**Pigpio version:**
$ pigpiod -v
79

**Node Version:**
node -v
v18.17.0

npm -v
9.6.7

**Package.json**
  "name": "RobotUI",
  "type": "module",
  "version": "1.0.0",
  "description": "Raspberry Pi FPV Robot",
  "main": "server.js",
  "dependencies": {
    "coffeescript": "^2.7.0",
    "express": "^4.18.2",
    "socket.io": "^4.7.1",
    "pigpio": "^3.3.1"
  },

This is the error that I am getting:
+-----------------------------------------------------------------------+
| Warning: The pigpio C library can't be loaded on this machine and any |
| attempt to use it will fail.                                          |
|                                                                       |
| Error: "Module did not self-register"                                 |
| ------------------------------------                                  |
| If you are working on a Raspberry Pi and see a "Module did not self-  |
| register" error, this typically indicates that the installation       |
| instructions were not exactly followed. For further details see the   |
| installation section of the readme at                                 |
| https://github.com/fivdi/pigpio#installation                          |
| Note that step 1 of the installation instructions must be completed   |
| before step 2.                                                        |
+-----------------------------------------------------------------------+

Invoking require('bindings')('pigpio.node') resulted in the follwoing error:

Error: Could not locate the bindings file. Tried:
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/build/pigpio.node
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/build/Debug/pigpio.node
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/build/Release/pigpio.node
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/out/Debug/pigpio.node
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/Debug/pigpio.node
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/out/Release/pigpio.node
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/Release/pigpio.node
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/build/default/pigpio.node
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/compiled/18.17.0/linux/arm64/pigpio.node
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/addon-build/release/install-root/pigpio.node
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/addon-build/debug/install-root/pigpio.node
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/addon-build/default/install-root/pigpio.node
 → /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/lib/binding/node-v108-linux-arm64/pigpio.node
    at bindings (/home/robot/Documents/Projects/javascript-robot-old/node_modules/bindings/bindings.js:126:9)
    at /home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/pigpio.js:8:31
    at Object.<anonymous> (/home/robot/Documents/Projects/javascript-robot-old/node_modules/pigpio/pigpio.js:43:3)
    at Module._compile (node:internal/modules/cjs/loader:1256:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1310:10)
    at Module.load (/home/robot/Documents/Projects/javascript-robot-old/node_modules/coffee-script/lib/coffee-script/register.js:45:36)
    at Function.Module._load (node:internal/modules/cjs/loader:960:12)
    at Module.require (node:internal/modules/cjs/loader:1143:19)
    at require (node:internal/modules/cjs/helpers:110:18)
    at Object.<anonymous> (/home/robot/Documents/Projects/javascript-robot-old/app.js:9:12)
    at Module._compile (node:internal/modules/cjs/loader:1256:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1310:10)
    at Module.load (node:internal/modules/cjs/loader:1119:32)
    at Function.Module._load (node:internal/modules/cjs/loader:960:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:81:12)
    at node:internal/main/run_main_module:23:47

basil-dsouza, 2023-07-29:
I just reinstalled again, but chose the 32bit version of bullseye, and this time it seems to have worked.
So it looks like this is a problem with 64bit os. Is that a known issue that I missed?

fivdi, 2023-07-30:
I can't reproduce the problem using the 64 bit version of Raspberry Pi OS.

With the 64 bit version of Raspberry Pi OS I can successfully install the pigpio Node.js module and successfully run the gpio-info example. This is what is see:

```
pi@raspberrypi:~ $ mkdir test
pi@raspberrypi:~ $ cd test/
pi@raspberrypi:~/test $ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
pi@raspberrypi:~/test $ pigpiod -v
79
pi@raspberrypi:~/test $ node -v
v18.17.0
pi@raspberrypi:~/test $ npm -v
9.6.7
pi@raspberrypi:~/test $ npm install pigpio

added 4 packages in 8s
pi@raspberrypi:~/test $ npm list --all
test@ /home/pi/test
└─┬ pigpio@3.3.1
  ├─┬ bindings@1.5.0
  │ └── file-uri-to-path@1.0.0
  └── nan@2.17.0

pi@raspberrypi:~/test $ cd node_modules/pigpio/example/
pi@raspberrypi:~/test/node_modules/pigpio/example $ sudo $(which node) gpio-info.js 
GPIO 0: mode=0 level=1
GPIO 1: mode=0 level=1
GPIO 2: mode=0 level=1
GPIO 3: mode=0 level=1
GPIO 4: mode=0 level=1
GPIO 5: mode=0 level=1
GPIO 6: mode=0 level=1
GPIO 7: mode=0 level=1
GPIO 8: mode=0 level=1
GPIO 9: mode=0 level=0
GPIO 10: mode=0 level=0
GPIO 11: mode=0 level=0
GPIO 12: mode=0 level=0
GPIO 13: mode=0 level=0
GPIO 14: mode=4 level=1
GPIO 15: mode=4 level=1
GPIO 16: mode=0 level=0
GPIO 17: mode=0 level=0
GPIO 18: mode=0 level=0
GPIO 19: mode=0 level=0
GPIO 20: mode=0 level=0
GPIO 21: mode=0 level=0
GPIO 22: mode=0 level=0
GPIO 23: mode=0 level=0
GPIO 24: mode=0 level=0
GPIO 25: mode=0 level=0
GPIO 26: mode=0 level=0
GPIO 27: mode=0 level=0
GPIO 28: mode=2 level=1
GPIO 29: mode=2 level=0
GPIO 30: mode=0 level=1
GPIO 31: mode=0 level=1
GPIO 32: mode=0 level=1
GPIO 33: mode=0 level=1
GPIO 34: mode=7 level=0
GPIO 35: mode=7 level=1
GPIO 36: mode=7 level=1
GPIO 37: mode=7 level=1
GPIO 38: mode=7 level=1
GPIO 39: mode=7 level=1
GPIO 40: mode=4 level=0
GPIO 41: mode=4 level=1
GPIO 42: mode=1 level=0
GPIO 43: mode=0 level=1
GPIO 44: mode=0 level=1
GPIO 45: mode=0 level=1
GPIO 46: mode=0 level=0
GPIO 47: mode=0 level=0
GPIO 48: mode=0 level=0
GPIO 49: mode=0 level=0
GPIO 50: mode=0 level=0
GPIO 51: mode=0 level=0
GPIO 52: mode=0 level=0
GPIO 53: mode=0 level=0
pi@raspberrypi:~/test/node_modules/pigpio/example $ 
```

You must have done something differently.

basil-dsouza, 2023-07-30:
Fair enough, I do not have the 64bit os any more, and since the 32 bit version is working well enough it no longer is an issue for me. Happy to close this issue. If I ever get back to it will comment here or raise a new issue.

closed 2023-07-30 by basil-dsouza