fivdi/pigpio #153: Simply not working

migoun, 2023-12-26:
This wrapper doesn't do anything. No error message or else, nothing happens. To me it looks like it cannot use the c-library at all.

Tested with Blinking LED example. I connected Pin 29. With the C-Library it works fine. Also on the terminal:
```
gpio mode 29 out
gpio write 29 1
```
Perfekt, led is on.

Only with the nodejs library, nothing happens at all.
I use this Code from https://docs.sunfounder.com/projects/davinci-kit/en/latest/nodejs/1.1.1_blinking_led_js.html

```
const Gpio = require('pigpio').Gpio;
const led = new Gpio(29,{mode: Gpio.OUTPUT});

var led_state = 0;

setInterval(() => {
  led.digitalWrite(led_state);
  led_state = !led_state;
  console.log(led_state);
}, 300);
```

Raspberry Pi 2, Model B

fivdi, 2023-12-26:
Just like the [pigpio C library](https://github.com/joan2937/pigpio), the pigpio Node.js module uses Broadcom GPIO numbers.

Please take a look at the image of the Pi GPIO header here https://www.raspberrypi-spy.co.uk/2012/06/simple-guide-to-the-rpi-gpio-header-and-pins/

If you are using pin 29 on the GPIO header, this means you are using GPIO5. If this is the case, your blink program should look like this:
```javascript
const Gpio = require('pigpio').Gpio;
const led = new Gpio(5,{mode: Gpio.OUTPUT});

var led_state = 0;

setInterval(() => {
  led.digitalWrite(led_state);
  led_state = !led_state;
  console.log(led_state);
}, 300);
```

If you are using pin 40 on the GPIO header, this means you are using GPIO21 (which, if I'm not mistaken, is pin 29 for Wiring Pi). If this is the case, your blink program should look like this:
```javascript
const Gpio = require('pigpio').Gpio;
const led = new Gpio(21,{mode: Gpio.OUTPUT});

var led_state = 0;

setInterval(() => {
  led.digitalWrite(led_state);
  led_state = !led_state;
  console.log(led_state);
}, 300);
```

fivdi, 2024-01-16:
I'll go ahead and close this as the Raspberry Pi GPIO header doesn't have a GPIO29.

closed 2024-01-16 by fivdi