fivdi/pigpio #154: Not working with lastest Raspberry pi OS 64bits (Bookworm)

lavoiekeven, 2024-05-04:
I had a projet with this library and It was working well last year. Now I'm setuping a new raspberry pi with Raspberry OS Lite 64 bits and I got this error:

Error: pigpio error -123 in gpioSetISRFunc
    at Gpio.enableInterrupt (/home/cdl/gateopener2/server/node_modules/pigpio/pigpio.js:264:12)
    at new Gpio (/home/cdl/gateopener2/server/node_modules/pigpio/pigpio.js:173:12)

It happen here:

`this._gpio = new Gpio(gpioPin, { mode: Gpio.INPUT, pullUpDown: Gpio.PUD_DOWN, edge: Gpio.EITHER_EDGE });`

I tried many pin all the same result.

I also start my app sudo.

Any ideas of what can be my problem?

lavoiekeven, 2024-05-05:
Ok by removing "edge: Gpio.EITHER_EDGE" it is working.

optikalefx, 2024-05-07:
@lavoiekeven could you share any sample code you're using? I'm trying to get a simple button working and I had the same error as you. The error goes away when making the change you suggested, but the button doesn't seem to do anything.

I'm doing this

```js
const Gpio = require('pigpio').Gpio;

const button = new Gpio(4, {
  mode: Gpio.INPUT,
  pullUpDown: Gpio.PUD_DOWN,
});

button.on('interrupt', (level) => {
  console.log('level', level);
});
```

Button is plugged into GPIO4 and GND

The log never happens

lavoiekeven, 2024-05-07:
Hi,
interrupt is not working ether.

So I used alert instead

exemple in TS

`import { Device } from "./Device";
const Gpio = require('pigpio').Gpio;

type AlertCallback = (level: 0 | 1) => void;

export class ContactSensor extends Device {
    private _gpioPin: any;
    private _gpio: any;
    private _alertCallback: AlertCallback;
    /**
     * 
     * @param {*} name 
     * @param {*} gpiopin 
     */
    constructor(name: string, gpioPin: number, callback: AlertCallback | null) {
        super(name, "sensor", false)
        this._gpioPin = gpioPin;
        this._gpio = new Gpio(gpioPin, { mode: Gpio.INPUT, pullUpDown: Gpio.PUD_DOWN });
        this._alertCallback = callback

        this._gpio.on('alert', (level) => {
            console.log(this.name + ' triggered');

            if (this._alertCallback)
                this._alertCallback(level);

            console.log(this.name + ' level changed ' + level);
        });

        this._gpio.enableAlert();
    }

    get gpioPin(): number {
        return this._gpioPin;
    }

    getLevel(): 0 | 1 {
        return this._gpio.digitalRead();
    }
}`

grevelle, 2024-06-22:
Is this project being maintained at this point?

lavoiekeven, 2024-06-22:
I guess not, but once i used alert rather than interrupt, the library work well in my case on. Personaly, I didn't find other library that worked  that well for what I wanted to do.

grevelle, 2024-06-22:
I am a huge fan of this library - it's the most powerful for hardware pwm IMHO. I use it extensively. The march bookworm update put me in a boot loop and I need to figure out the problem. I use alerts in my code already so glad to hear that's not the problem.

grevelle, 2024-09-05:
I am having the same issue w/ interrupt vs. alert. I can get alerts to work but not interrupts.