fivdi/pigpio #155: pigpio : Error -123 gpioSetISRFunc

TMagerat, 2024-08-05:
Hello

I'm struggling with the function
gpioSetISRFunc
I always have a bad init error -123
I tried to find some solution by disabling the W-1 interface but it does not work 

This is my config.txt

I need this function because I have 2 different square signals on my PIN BCM 17 and 27 that represent the speed of 2 different rotors
I want that each time a RISING_EDGE is occuring it computes the times between the last and the current rising_edge

I tried to do 

```
sudo -s
cd /sys/class/gpio/
echo 17 > export 
cd gpio17
echo in > direction 
echo both > edge 
cd /sys/class/gpio/
echo 17 > unexport 
```

as mentionned here #97  https://github.com/fivdi/pigpio/issues/97

but sadly after the line 
```
echo 17 > export 
```
I have an invalid argument error
My code is something like this :
```
void rotor1_interrupt(int gpio, int level, uint32_t tick) {
    static uint32_t last_tick = 0;
    uint32_t diff_tick;

    
    diff_tick = tick - last_tick;
    last_tick = tick;

    if (diff_tick > 0) {
        rotor1_speed = 1.0 / ((double) diff_tick / 1000000.0);
        clock_gettime(CLOCK_MONOTONIC, &speed1);
        time1 = (speed1.tv_sec - start.tv_sec) + (speed1.tv_nsec - start.tv_nsec) / 1.0e9;
    }
    rotor1_speed_buffer[rotor1_data_index] = rotor1_speed;
    time1_buffer[rotor1_data_index] = time1;
    speed_order_buffer[rotor1_data_index]=speed_order;
    rotor1_data_index++;

    update_pwm_flag_rotor1 = 1; // Indicate that rotor 1s speed has been updated
    
}
...
int main(int argc, char **argv) {
    ...
    if (gpioInitialise() < 0) {
        fprintf(stderr, "Pigpio initialization failed\n");
        return 1;
    }
    gpioSetMode(ROTOR1_PIN, PI_INPUT);
    gpioSetMode(ROTOR2_PIN, PI_INPUT);
    gpioSetMode(PWM_PIN, PI_OUTPUT);

    gpioSetISRFunc(ROTOR1_PIN, RISING_EDGE, 0, rotor1_interrupt);
    gpioSetISRFunc(ROTOR2_PIN, RISING_EDGE, 0, rotor2_interrupt);
    ...
    }
```

It fails at the first gpioSetISRFunc

Thanks in advance 

Tom

Crylion, 2024-08-06:
Having the same issue :'(

TMagerat, 2024-08-06:
> Having the same issue :'( My Pi is on Debian 12, more or less freshly installed and 1-wire is definitely not enabled

Hello,

I found the function 
gpioSetAlertFunc( gpio_pin , interrupt_function); in the library pigpio
It is exacly the same, but it detects every change of state of your gpio.
However, in your interrupt function, you have access the an int named "Level" which is describing if it is a rising_edge or a 
falling_edge 

so you do the following code: 

void interrupt_function (int gpio, int level, uint23 tick){
   if (level==1){
      Code_of_your_previous_interrupt_function;
}

Rising_edge : level ==1

I hope it was helpful for you